DiedB/Homey-SolarPanels #1: "This app has crashed"

Dijker, 2016-05-16:
**What is your Homey version (Settings → System)?**
0.8.34 / App v1.2.0

**What did you try to achieve?**
App ran approx. 18 days (with a lot of restarts for Homey .32 / KAKU etc.) without problems monitoring my SolarEdge. 
Today the Settings / Apps showed an exclamation mark "This app has crashed" (ChromeCast also) 
Graphs updating stopped yesterday. 

*Please include detailed steps how we can reproduce the issue*
Don't know yet if I can reproduce!

**What was the result?**
Graphs updating stopped yesterday. 

**What did you expect as result?**
No Crash ;-)

**Any other remarks**
Disabled and re-enabled the App 2 times, Solar Panels works some time but crashes again.
Don't know if .34 is unstable or the App,  or the missing restarts of Homey ;-)

DiedB, 2016-05-17:
I'll try to reproduce it, I have installed 1.2 with a SolarEdge and I'll see what happens.

DiedB, 2016-05-23:
How is it working with .35?

I do get some crash reports in my e-mail, these are always related to connection issues (internet connection down or API down). I will solve this in the next version.

Dijker, 2016-05-27:
Today again 
The exclamation mark pipes up the following message:
Hope you know what it means  :-)

Stack trace:

TypeError: Cannot read property 'statusCode' of undefined
at Request._callback (/drivers/solaredge/driver.js:122:49)
at self.callback (/node_modules/request/request.js:197:22)
at emitOne (events.js:77:13)
at Request.emit (events.js:169:7)
at Request.onRequestError (/node_modules/request/request.js:854:8)
at emitOne (events.js:77:13)
at ClientRequest.emit (events.js:169:7)
at TLSSocket.socketErrorListener (_http_client.js:256:9)
at emitOne (events.js:77:13)
at TLSSocket.emit (events.js:169:7)

DiedB, 2016-06-06:
It is probably your internet/WiFi connection dropping out for a minute, or it is the API dropping out. In the next version this error will be handled (the system will be put on 'offline' if the status can't be fetched).

DiedB, 2016-06-06:
New version will be out in a few hours, I will mark this as solved for now.

If problems still occur, we will notice (I get crash reports in my e-mail too), and I'll reopen the issue.

closed 2016-06-06 by DiedB