DiedB/Homey-SolarPanels #2: Cannot connect

edtst, 2016-05-19:
I cannot get my system online in the app. 

System ID: Is this the last numerical part of the auth url?
API: I filled out the API key of my app in the developer area of enlighten.
User ID: Is this the user ID of the developer site?
Name: I chose a random name.

DiedB, 2016-05-19:
What system are you trying to connect?

edtst, 2016-05-20:
It is an enphase system with an Envoy S.

DiedB, 2016-05-20:
Have you authorized your API account to get your system data? Follow the
instructions in the Enphase portal.

From there on, the README should be self-explanatory. If you need help, let
me know!

edtst, 2016-05-20:
Yes, done that. I just don't know where to het the system ID to enter in Homey.

DiedB, 2016-05-20:
Could you try following this guide:
http://www2.enphase.com/enlighten-help/tip/where-do-i-find-my-user-id-and-the-system-site-id/

Let me know if that works. If it doesn't, please send me the details you
have via e-mail, and I'll try to reproduce the error to see which of your
parameters is incorrect.

edtst, 2016-05-20:
I tried that. You need to have a supplier account to access that page. DO you know what this id looks like? Is it a number?

DiedB, 2016-05-20:
It is a number, 5 or 6 digits if I recall correctly.

edtst, 2016-05-20:
I think I found it, but it keeps telling me error 401 not authorized. Could you give me the parameters you used for the demo system?

DiedB, 2016-05-20:
Yes, but I prefer not to post them here since it's my own API key. Do you have an e-mail address?

edtst, 2016-05-21:
Yes, but how to send it. Is the e-mailadres in your github code real?

DiedB, 2016-05-21:
Don't know what you mean by GitHub code, but it probably is.

edtst, 2016-05-22:
I managed to find the correct system id, it was in the url and matches one that I see in a mobile app. It is, as you said, the 6 figure number. I also have the API key and authorized the app. My enlighten access tells me the authorization succeeded and I can see the developers app I created on the developer site. I also used the user ID I was given when authorizing.

Still the app in homey gives me Not authorized to access requested source

edtst, 2016-06-06:
I found out what the problem was. I needed to enter the user ID and systrem ID manually and not copy it from the webpage from Enlighten.

closed 2016-06-06 by edtst