DiedB/Homey-SolarPanels #4: SolarEdge API not working anymore

mruiter, 2016-11-25:
Hello,

I just noticed my SolarEdge API connection is not working anymore.
Doublechecked the system id and api  

Deleted the device and recreated it with the right settings, al stays on 0 0 
Running on 1.0.1

DiedB, 2016-11-29:
Can you try completely reinstalling the app and then adding the device again?

If that doesn't work, please send me your API parameters and I'll start troubleshooting.

Dijker, 2017-01-15:
Same problem here, 
Solaredge website still works
Domoticz still works with same keys
Production Homey stopped working last month .... eventually removed App and re-installed breaking several flows and losing all history (Yea I was aware of that... a nice backup/restore would be nice ;-) )
Solaradge connection Still doesn't work. 
My Second Dev Homey has the same, Pairing Ok, but get only 0 Watt / 0 Wh  and no new data. 

@DiedB  Do you already have a key to test?

DiedB, 2017-01-15:
I have the key to test it. Now onto having some time to fix it! I have some things on the list for 1.4.2, this bug is one of them as well as some new inverters, but I can't give a date yet.

DiedB, 2017-01-21:
The bug has been smashed. Registering cron after a reboot didn't happen if the task already existed.

closed 2017-01-21 by DiedB