DiedB/Homey-SolarPanels #5: SAJ not showing energy and power

chrisboer, 2017-03-21:
Homey v1.2 RC8
SAJ not showing energy and power readings anymore ?

Had to remove them and set them up again.

Readings from Solarman keeps on working fine

DiedB, 2017-03-21:
How long did it work before? When did it stop (you can see it in insights)? Did anything happen to Homey at that moment?

chrisboer, 2017-03-21:
It worked untill I updated my homey with v1.2 RC8 so after that it stopped working for SAJ

DiedB, 2017-03-21:
Hmm, strange. My inverters also kept working through this update.

I am waiting for feedback from other SAJ users. Does the inverter keep working after Homey reboots and other updates?

chrisboer, 2017-03-21:
Works perfect again also after multiple reboots
[image]

So you have to re-assign them problem solved

closed 2017-03-21 by chrisboer