DiedB/Homey-SolarPanels #6: SAJ energy readings not visable

chrisboer, 2017-03-27:
Homey 1.2.0 RC9 
Solar inverters from SAJ both no readings recorded anymore. So again an issue probably after a reboot or system disconnect from homey. 

[image]

DiedB, 2017-03-28:
Are you sure the SAJ inverter is reachable? If you enter the IP address of the inverter in a browser, do you see the live data?

chrisboer, 2017-03-29:
Yes I am sure they were reachable, I have now given them a fixed IP address because with the installation of 1.2.0 RC8 the SAJ were not giving any info and SAJ were working but not on homey.

chrisboer, 2017-03-29:
Updated to RC10 tonight let's see if they still work tomorrow.
[image]

chrisboer, 2017-03-30:
Updated to 1.2.0 RC10 and view reboots and disconnects and new day and it is still working.

closed 2017-03-30 by chrisboer