DiedB/Homey-SolarPanels #9: Ginlong connected but no data

sleepmaker, 2018-03-16:
I've connected my Homey to my solarpanels, using the login credentials from the ginglong monitoring website. The credentials are accepted, I copied the plant-ID from what was shown in the browser bar after logging in. So far everything seems fine. But..... I can't see any energy production being recorded.
When viewing through the dedicated ginglong-app on my phone I can see the energy production.
Why is the Homey plugin not recording data?

DiedB, 2018-03-19:
There is a reason I didn't include Ginlong in the current beta version, but I don't remember exactly. I will update this app in the summer (when I have got the time), then I'll look into Ginlong support again.

You're always welcome to submit a pull request!

snelis, 2018-04-11:
It works for me, just installed it - did the same - also copied the plant id from the url like pid=1337

sleepmaker, 2018-04-11:
@snelis so you are using an older version, is that right?

snelis, 2018-04-11:
v1.4.3 of the app in homey

sleepmaker, 2018-04-11:
I have the same. Then why the .... is it not working at my place?

snelis, 2018-04-11:
so if you go to homey insights, and select the solarpanels u see no data points ?

sleepmaker, 2018-04-11:
Nope. Not a single one. And I just checked and rechecked my plantID. My energy suppliëren show my plant production, so that is Working. I just don’t get it. Maybe I need some help from the technical Guys out there ....

DiedB, 2018-04-12:
I will take a look into this some time in the next few weeks. Do you have an account for me to test? Just send it via e-mail or private message (if that is a thing on GitHub).

DiedB, 2018-04-18:
Ginlong, Omnik, Trannergy and Solarman all use the same API framework. The last three migrated to a new version, while Ginlong did not. I just manually tried approaching the Ginlong API, and logging in works but fetching data does not give any result. Their app probably has the same problem, could you try installing that (on Android/iOS)?

This is also the reason why I didn't migrate the Ginlong driver to the new version of this app.

sleepmaker, 2018-04-18:
I have the app on my iPhone. But this one does give me live output. As you may have noticed it is quite a sunny day today and my panels are producing 4.71 kW right now as I check the app

DiedB, 2018-04-18:
I'll reverse engineer the app again, in that case. It might use a different endpoint or API key in that case.

sleepmaker, 2019-01-18:
OK. So Solis support is completely dropped as of the newest version?

DiedB, 2019-01-19:
It is not supported in the current version, as they do not provide a documented API.

leo389, 2019-03-30:
Hello,

Thanks for you effort in developing the app.

I've got a Solis inverter with the following wifi dongle:

https://www.cclcomponents.com/solis-plug-in-wifi-stick

I note that in the dongle interface it's possible to setup 3 remote servers (one of which I assume is the Ginlong site). The page states the following:

> Your device can transfer data to several servers in the form of TCP/UDP if your device supports this function.
> Please make sure that the response to ping request function is enabled on the assigned server for receiving data.
> 

Do you think it would be possible to use the functionality to send the data to Homey? I can assist with development if it helps.

Regards,

DiedB, 2019-04-01:
Their monitoring platform does not seem to have an API.

Maybe a direct connection to the inverter could be an option, but that is very hard to test if I do not have one of these devices in my own local network.

https://github.com/graham0/ginlong-wifi/blob/master/read-ginlong.py

If someone wants to look into this and create a pull request, feel free!

leo389, 2019-04-01:
I might get a chance in the next week or two to try the above script on a Linux system but will Homey accept such data on an incoming port?

I can setup my system to point to a particular server if that helps with development.

sleepmaker, 2019-04-01:
Nice work fella’s. I don’t have the programming knowledge to help you out.
But it would be greatly appreciated if you can get this to work

If there’s anything I can assist you with, please let me know

Thom

DiedB, 2019-04-01:
@leo389 Ah, I see that the script opens up a server that listens for data from the inverter stick. That will probably not be possible using the current Homey SDK and is also a little too complicated for the app.

leo389, 2019-04-01:
Bit of a pity as I just tested the script & it works as advertised!

DiedB, 2019-04-02:
Do you have a PCAP file?

leo389, 2019-04-03:
Don't have a pcap, will try & grab one but the request from the app looks
straightforward enough, a get request with a user ID & a token. The return
is all the status info.

Will try & see if I can replicate a request.

leo389, 2019-04-03:
Some success it seems!

I've managed to retrieve data using the following curl command (from a linux system):

curl 'http://apic-cdn.solarman.cn/v/ap.2.0/plant/get_plant_overview?uid=123456&plant_id=12345'

The pieces of info needed are straightforward to obtain:
1. uid - after logging into https://m.ginlong.com/main.html at top right hand corner (user ID)
2. plant_id- after logging into https://m.ginlong.com/main.html -> plant info

Initially I thought a token was required but so far it seems to work without one. If the token is required I obtained mine by using a packet capture app on my phone to capture packets & picked the token from that. Hopefully it continues to work without the token.

I can send on the retrieved data if you like.

leo389, 2019-04-13:
Whats the best way to help you reincorporate the Ginlong inverters? Is it to try & modify something like 

https://github.com/DiedB/SolarPanels/tree/ca96283269138fb0e372e3c9369ebbd659ea4cde/drivers/solaredge

It seems to be implemented with parsing a json stream.

DiedB, 2019-04-15:
If you look at the alpha branch, the Enphase implementation I just pushed is a great example. Let me know if you have any questions and I'll be happy to help!

leo389, 2019-04-15:
Just created an upload request with the work I've done. It's just a start but some work done. I'm unsure of some of the nodejs stuff (I use mostly C) so it's rough around the edges.. 

The app builds & runs but's getting stuck in api.js.

Also I'm unsure if I created the pull request properly..might be the wrong branch

DiedB, 2019-04-16:
Cool! I have removed and recreated the PR in the correct way, you can find it here: https://github.com/DiedB/SolarPanels/pull/55/files

I cannot give you access to push to the branch though, you will have to create a fork with your changes and then submit a pull request to allow that.

Can you share your Ginlong account with me for testing? My email address is diederik@outlook.com

DiedB, 2019-04-17:
I've built an implementation that runs fine. I do not trust this API one bit however. Invalid JSON, misspelled names, Chinese signs everywhere. We'll just see if it keeps working.

I'll add the two people that sent me an email with account details to the alpha version. Please report back if it's working for you.

sleepmaker, 2019-04-17:
Bummer.
Installed the alpha. Thanx for the invite btw.
I’m getting an error message. It’ll be attached to this mail.
Hope it helps

Thom

Verstuurd vanaf mijn ZX-81

sleepmaker, 2019-04-17:
It looks like the plantID has changed
Using a new one (163753) the app goes on a little further but then another error message appears

Verstuurd vanaf mijn ZX-81

sleepmaker, 2019-04-17:
.......
Nevermind
My plant id is somewhere else on the new website to be found (128306)
This one works! For now

Verstuurd vanaf mijn ZX-81

DiedB, 2019-04-17:
Can you document the process of finding the plant ID? I would like to include it in a README before publishing.

sleepmaker, 2019-04-17:
Login to "http://m.ginlong.com/main.html" using your Ginlong credentials
Select "installatie informatie" in the upper part of the screen
On the new screen you will find your "installatie-ID"
Use this number as your "Ginlong PLANT-ID" in the Homey-app

Does this suffice?

Thom

leo389, 2019-04-17:
Many thanks for the work, I hope my stuff was of some use!

v3.0.5 is working for me, & the data it presents matches the Ginlong app. 

If I may could I suggest the following editing/ addition to sleepmakers process

1. Login to "http://m.ginlong.com/main.html" using your Ginlong credentials 
2. Select "Plant Info/ Installatie informatie/ " in the upper part of the screen 
3. On the new screen you will find your "Plant ID/ Installatie-ID/ " 
4. Use this number as your "Ginlong PLANT-ID" in the Homey-app 

Would it be possible to include in the app some method of indicating that the API has broken? Perhaps push an Issue to this github repo or notifying the user on their phone?

sleepmaker, 2019-04-17:
I’m OK with this

DiedB, 2019-04-19:
Has been released in v3.0.6. Forgot adding the guide, will do in a future release. Thanks for the help!

closed 2019-04-19 by DiedB