DiedB/Homey-SolarPanels #10: Can't connect to SolarMAN

RonaldDZ, 2018-04-17:
Ik heb SolarPanels (v1.4.3 -  12.06.17) geïnstalleerd en heb geprobeerd een SolarMAN apparaat toe te voegen. Nadat ik de gevraagde gegevens (username, password, site ID en inverter naam) had ingevuld en op 'Connect' klikte kreeg ik de foutmelding 'Plant ID onjuist'.
De ingevulde gegevens zijn correct en ik kan mijn SolarMAN account via de SolarMAN website zonder problemen benaderen.
Ik heb ook de beta-versie (v2.2.1 -  25.03.18) geprobeerd. Bij deze versie verschijnt na het invullen van de gegevens en het klikken op 'Connect' een wachtsymbool (draaiend tandwiel) en een button 'Volgende'. 
Het wachtsymbool verdwijnt niet vanzelf, en als ik op 'Volgende' klik sluit het venster maar is er geen apparaat toegevoegd.
Wat kan hiervan de oorzaak zijn?

DiedB, 2018-04-18:
Is your plant ID a 5-digit number?

I suggest using the beta, by the way, it is more robust 😄

RonaldDZ, 2018-04-24:
The plant ID was alphanumerical, I changed it to a 5-digit number but that didn’t do the trick.

I tried both versions of the app, but with neither of them I could connect tot he SolarMAN server.

DiedB, 2018-04-30:
Are you retrieving the plant ID from the URL after logging in? Just changing it to a random 5-digit number will not do the trick.

RonaldDZ, 2018-04-30:
On the SolarMAN site you have to set a plant ID yourself, it’s not defined by SolarMAN.

RonaldDZ, 2018-04-30:
I understood it when I was reading the other issue abou Ginlong, it's working now :-)

closed 2018-04-30 by DiedB