DiedB/Homey-SolarPanels #15: Goodwe can't connect anymore

thezahir1, 2018-06-04:
- Since a few days, (around June 1) the app reports 302 for my Goodwe inverters. Hopefully it can be fixed?
- After updating to the beta version (2.2.x) it seems that Goodwe support has disappeared, is that correct? Will it stay that way...?

epjjansen, 2018-06-04:
I have the same issue.
I had a message from the Iphone app that the previous built of the app was unsupported. Only 3.0.7 will work. So I think this has something to do with the 302 error.

DiedB, 2018-06-04:
Sounds like they upgraded their API. Got the same signal from three users in one day, so I will look into it as soon as possible. That will also mean support for GoodWe will be moved to the beta channel. Stay tuned for updates.

epjjansen, 2018-06-04:
I think this has to do with the deployment of a new European server and the GDPR law.

Asinke, 2018-06-10:
Same issue here.
@DiedB: Looking forward for an update

DiedB, 2018-06-10:
An update has been released on the alpha channel, feel free to manually download and run that version (alpha branch on Git).

I will test it and push it to the beta channel tomorrow.

Asinke, 2018-06-10:
Excellent community support!
Installed the Alpha version as indicated and now the GoodWe reporting works again!
Only thing I noticed is that no password for the Goodwe? account was requested?

Thanks,

Arthur

DiedB, 2018-06-10:
GoodWe and security do not go hand in hand apparently, their APIs are completely opened up for everyone.

thezahir1, 2018-06-10:
Good busy, @DiedB! When will it be available thru normal download in Athom's appstore? (I prefer to install it the normal way)

DiedB, 2018-06-10:
As it seems to be working fine for multiple users, I have submitted it for review by Athom. It will show up in the beta channel soon.

closed 2018-06-10 by DiedB