DiedB/Homey-SolarPanels #17: SAJ connection fails

HansWaldram, 2018-06-11:
I'm trying to connect my SAJ converter but keep getting the message "Incorrect IP address". Is there a known issue?

DiedB, 2018-06-11:
Not yet! What do you get if you access the IP address directly?

HansWaldram, 2018-06-11:
When I access the IP directly I get the status page of my SAJ converter.

DiedB, 2018-06-11:
Is Homey on the same network? Do not put http:// in front or a / at the end.

HansWaldram, 2018-06-11:
Homey is on the same network. I put the IP as 10.10.0.127 

So these should be OK.

DiedB, 2018-06-12:
Could you try going to 10.10.0.127/real_time_data.xml (fill in the real IP). Does it yield any data? If so, could you post the full XML in a code block here?

HansWaldram, 2018-06-18:
Sorry voor de vertraging, ik was even in het buitenland...

Daar komt de melding 404: The requested file cannot be found.
De foutmelding waar het mee begon is hier te openen:
[image]

HansWaldram, 2018-06-18:
En hier de settings van de converter:
[image]

DiedB, 2018-06-19:
What inverter model do you have?

HansWaldram, 2018-06-21:
A Sununo plus 5k-m.

DiedB, 2018-06-22:
Thanks! Did some research, some inverters have a different interface (like yours) than others. [This](https://www.domoticz.com/forum/viewtopic.php?p=176482&sid=d94263082784f88e73e6522d6a289245#p176482) forum thread describes the issue and the alternative way of fetching data for those inverter types.

I will try to create a fix for this in the summer. Will probably work by detecting the type of inverter.

HansWaldram, 2018-07-20:
Hi Diederik, great. I’ll wait for your update and will be happy to test once you’re ready.

HansWaldram, 2018-08-28:
Hi Diederik, have you had the chance to look into this matter already?

DiedB, 2018-08-28:
Unfortunately, not yet. Due to other activities, I am afraid I will be unable to do anything on this app until November.

HansWaldram, 2019-01-05:
Hi Diederik, best wishes for 2019!

Any update on the above?

HansWaldram, 2019-01-12:
Hello Diederik, what is your progress on the above?

HansWaldram, 2019-02-13:
Hi Diederik, are you no longer supporting the project?

DiedB, 2019-02-14:
Hi! I have lots of things to do and the maintenance on this app is not my top priority, unfortunately. Also, I want to focus on getting inverter brands with clear API documentation and support fully working first, which is why other issues are taking precedence.

HansWaldram, 2019-02-14:
Hi Diederik, apart from that you indicated November I understand. Any indication if and when you could work on the issue?

DiedB, 2019-04-12:
Can you try and visit the following page (fill in username, password and IP):

http://<LOGIN_INVERTER>:<PASSWORD_INVERTER>@<IP_INVERTER>/status/status.php

And paste what you see there?

DiedB, 2019-04-19:
Closing because of lack of activity, please re-open if needed.

closed 2019-04-19 by DiedB

HansWaldram, 2019-04-20:
I’ll get back to you upon my return from holiday.

Groet,
Hans

telenut, 2020-11-28:
please re-open... same problem here. 
Just installed solar panels. Looked up first if SAJ was suported by homey, and then you get this :s

telenut, 2020-11-29:
To answer that question; this is what you get on that url=
1,370,62,0,4,2044,3,1713,1,65535,65535,0,0,0,0,0,0,0,0,0,0,0,0,5,4999,2328,8,65535,65535,65535,65535,3548,194,29,2