DiedB/Homey-SolarPanels #18: Solax inverter

TijnBe, 2018-07-16:
I have a Homey and a Solax inverter ZDNY-TLX3 and would like to use it through Homey. Do you have any plans for the Solax?

Groet
Tijn

DiedB, 2018-07-17:
There sure are! A pull request (#16) has been submitted for implementing Solax. Could you help testing it with your system? Do you know how to run apps in debug mode on Homey?

TijnBe, 2018-07-17:
Hi Diederik,
Leuk! Ik weet niet hoe ik in de debug mode kan maar dat zal vast niet ingewikkeld zijn. Prima om mijn systeem te gebruiken. Wat heb je nodig?

Verstuurd vanaf mijn iPhone
Goede groet
Tijn Bettink

TijnBe, 2018-07-18:
I am not in to programming. I have some technical knowledge and would like to try. Can you tell me step by step what to do. So I can clone.. For your info I know Solax is working with two different websites for the info. solax-portal.com and www.solaxcloud.com (the one I use)

DiedB, 2018-07-18:
Install NodeJS (LTS) and Git.

`git clone https://github.com/DiedB/SolarPanels.git`
`git checkout -b Willybits-master alpha`
`git pull https://github.com/Willybits/SolarPanels.git master`

Then follow the steps in the Athom guide I referred to in my previous comment, except for the fourth one.

TijnBe, 2018-07-18:
I get to this

Last login: Wed Jul 18 11:44:32 on ttys000
MACBOOK-TIJN:~ admin$ cd solarpanels-master
MACBOOK-TIJN:solarpanels-master admin$ npm install athom-cli
npm WARN SolarPanels@ No license field.

+ athom-cli@2.1.23
updated 1 package in 3.634s
MACBOOK-TIJN:solarpanels-master admin$ athom login
-bash: athom: command not found

DiedB, 2018-07-18:
`npm install -g athom-cli` should work!

TijnBe, 2018-07-18:
+ athom-cli@2.1.23
updated 1 package in 8.696s
MACBOOK-TIJN:solarpanels-master admin$ npm install -g athom-cli
npm WARN checkPermissions Missing write access to /usr/local/lib/node_modules
npm ERR! path /usr/local/lib/node_modules
npm ERR! code EACCES
npm ERR! errno -13
npm ERR! syscall access
npm ERR! Error: EACCES: permission denied, access '/usr/local/lib/node_modules'
npm ERR!  { Error: EACCES: permission denied, access '/usr/local/lib/node_modules'
npm ERR!   stack: 'Error: EACCES: permission denied, access \'/usr/local/lib/node_modules\'',
npm ERR!   errno: -13,
npm ERR!   code: 'EACCES',
npm ERR!   syscall: 'access',
npm ERR!   path: '/usr/local/lib/node_modules' }
npm ERR! 
npm ERR! Please try running this command again as root/Administrator.

npm ERR! A complete log of this run can be found in:
npm ERR!     /Users/admin/.npm/_logs/2018-07-18T10_17_37_050Z-debug.log
MACBOOK-TIJN:solarpanels-master admin$

DiedB, 2018-07-18:
Never mind, I read that wrong.

Easiest option then is running `athom-cli` locally. `$(npm bin)/athom login`?

TijnBe, 2018-07-18:
Locally what do you mean by that? And what with $(npm bin)/athom login ?
thanks

DiedB, 2018-07-18:
You first installed athom-cli locally, by using `npm install athom-cli`. Then globally, using the `-g` flag. When installed globally, you can use the command everywhere, but there is some issue with permissions which is probably not present in the local version.

By using the command `$(npm bin)/athom login` in the SolarPanels folder, you should be able to use the Athom CLI.

TijnBe, 2018-07-18:
This worked after searching the internet and I am in..
**sudo chown -R $(whoami) /usr/local/lib/node_modules**

Skip step 4 but what to do at step 5 not clear to me

TijnBe, 2018-07-18:
nou ook opgelost volgens mij heb ik alles nu gedaan....

DiedB, 2018-07-18:
So, the Solar Panels app is installed on your Homey? Can you add your Solax inverter now?

TijnBe, 2018-07-19:
Yes just installed and no the Solax is not visible.

DiedB, 2018-07-19:
As in, you cannot choose Solax in the pairing wizard?

TijnBe, 2018-07-19:
Installed version v2.1.0
I can choose the next brands
PVoutput, Solaredge,enphase,omnik,tigo,solarman,trannergy,sungrow

[image]
[image]

TijnBe, 2018-07-22:
I removed the complete folder an reinstalled it again. No difference. This is what I get
macbook-tijn:solarpanels-master admin$ git status
fatal: not a git repository (or any of the parent directories): .git
macbook-tijn:solarpanels-master admin$

TijnBe, 2018-07-22:
I reinstalled again and now it is working i can choose the Solax. No idee why it is possible now.
You have to give 'gebruikersnaam', 'wachtwoord', 'site-id' for the portal https://www.solax-portal.com/. This portal is working for some Solax inverters and for others it is https://www.solaxcloud.com/ This is different and not working with an site-ID.

DiedB, 2018-07-24:
Can you send me your login details (d@iederik.nl)? I will look into implementing the alternative Solax portal.

Willybits, 2018-07-29:
Hi Tijn,
What Inverter do you have? (type) ?

TijnBe, 2018-07-30:
Hi Willybits,

I have the ZDNY-TLX3-7.0. What they told me at Solax is that the solaxcloud is there own portal and that solax-portal.com is not. The wifi module I have is version 2.0. I had first the 1.0 version but that did not work with the inverter no data was transmitted. It seems that the 1.0 is working with solax-portal.com and the 2.0 with the solaxcloud.com

I don't know which inverters are working with which portal.
Curious what they will respond.

Willybits, 2018-07-30:
Hey, Thanks for the info.

Aha, so for now it's safe to assume that WIFI version 1 uses Solax-Portal and wifi version 2 uses solaxcloud. Thats also the reason then why I cannot add my system as my SN from the WIFI module is recognised as a V1 module.

I hope they will reply as well to tell what their plans are.

For Solax support in Homey this would mean we will have 2 portals to support.

DiedB, 2018-07-30:
I think we will need to support both portals, indeed. The question that currently needs answering is whether Solax Cloud has an API, and action is being taken into getting the documentation if there is. Keep me up to date about the progress.

TijnBe, 2018-08-01:
I have direct response from Solax. Diederik is in cc.

Willybits, 2018-08-01:
Could you post the resonse here as well, I'm curious :)

TijnBe, 2018-08-01:
Wat is de communicatie methode.
Welke informatie wil je uitlezen van de omvormer en hoe. (bekabeld wifi, of wil je de informatie via het portaal uitlezen.)

legalr, 2018-08-21:
Hello
i am curious, what is the status of implementing the SolaxCloud website.
I just installed a Solax inverter connected to the solax cloud website, and i like to see it working with the Homey App.

DiedB, 2018-08-21:
Just sent Solax a reminder, I need their documentation to do anything.

TijnBe, 2018-09-12:
Already some reaction from Solax?

DiedB, 2018-09-12:
Nothing, unfortunately. I'll be abroad for the next 8 weeks so I will be unable to do any development on this app.

bossan, 2019-01-09:
Hello all,
I just got it working to retrieve data from SolaxCloud.
I did it in Python, here is my code:

```
            # Starting a session in which we will store a JSESSIONID Cookie
            s = requests.Session()
            # Logging in to get the Cookie
            s.post(
                url='https://solaxcloud.com/login/login.do', 
                data={
                    "username": self._username, 
                    "userpwd": self._password, 
                    "roletype": 5
                }, 
                verify=False
            )
            # We can now make a request to get all data from a specific date
            r = s.post(
                url='https://solaxcloud.com/export/getReportData.do', 
                data={
                    "type": 1,
                    "siteId": int(self._siteId),
                    "time": date_str
                },
                verify=False
            )

            data = r.json()
```

The result will look like this:
```
[
{
        "inverterSn": "xxxxxxxxxx",
        "wsn": "xxxxxxxxxxx",
        "moduleName": "xxxxxxx",
        "userId": "xxxxxxxxxx",
        "siteId": "xxxxxxxxxxx",
        "firmId": "xxxxxxxxxxx",
        "firmwareVer": "7",
        "ratedPower": 5,
        "idc1": 0.2,
        "idc2": 0.4,
        "idc3": null,
        "idc4": null,
        "vdc1": 236.5,
        "vdc2": 292.8,
        "vdc3": null,
        "vdc4": null,
        "iac1": 0.6,
        "vac1": 230.4,
        "gridpower": 136,
        "temperature": 27,
        "todayyield": null,
        "yieldtoday": 0,
        "yieldtotal": 24.2,
        "feedinpower": 0,
        "powerdc1": 45,
        "powerdc2": 91,
        "powerdc3": null,
        "powerdc4": null,
        "pac1": 47,
        "pac2": 43,
        "pac3": 46,
        "iac2": 0.6,
        "iac3": 0.6,
        "vac2": 230.9,
        "vac3": 225.6,
[... 130 lines cut ...]
        "batteryCellVoltageHigh": 0,
        "batteryCellVoltageLow": 0,
        "batteryHealth": "0",
        "bmscommunication": "0",
        "minCapacity": 0,
        "userChargemaxCurrent": 0,
        "userDischargemaxCurrent": 0,
        "workMode": 0,
        "inverterError": 0,
        "bmserror": 0,
        "managerRrror": 0,
        "bmschargemaxCurrent": 0,
        "meterFunction": 0,
        "meter1ComState": 0,
        "meter2ComState": 0
    },
   {...}
]
```

I only use the value "gridpower" which is the total amount of generated power at that point in time (in Watt).

Please note, I had some problems with verifying their SSL Certificates, hence the `verify=False`.

I hope it answered your question :)

DiedB, 2019-01-14:
We could do this, but it is not as robust as using an API. No one knows when they will change their login handler, and I do not have time to fix constant issues with the app breaking.

loopezz, 2019-05-25:
Is there any progress for Solax Cloud support? I'm willing to help with testing

DiedB, 2019-05-27:
I will not have time for this in the foreseeable future. Maybe @bossan wants to try and create an implementation for Homey?

bossan, 2019-05-27:
@loopezz @DiedB Unfortunately I do not own a Homey device, so I am not sure if I will be able to add support for it.

magcb, 2019-08-03:
Hi, can you please add Solax cloud support in the app?
Thanks

Willybits, 2019-10-24:
Could you all try this on your Solax inverters and paste the result?
http://INVERTERIP/api/realTimeData.htm

The result should be something like:
{“method”:“uploadsn”,“version”:“Solax_SI_CH_2nd_00000000_DE02”,“type”:“AL_SE”,“SN”:“12345678”,“Data”:[0.0,0.0,0.0,0.0,1.8,228.9,348,34,0.6,31.5,-40,0,0,51.87,-7.99,-416,19,58,0.0,10.9,0.00,0.00,50.01,0.0,0.0,0,0.00,0,0,0,0.00,0,8,0,0,0.00,0,8],“Status”:“2”}

Curious if all solax inverters present their data there. Mine does, so perhaps its easier to use the local API.

oliviervb, 2020-02-12:
hello any progress?

DiedB, 2020-02-20:
@oliviervb I do not have enough data or documentation to add support currently, but maybe you can try the suggestion of @Willybits if you want support for it?

oliviervb, 2020-02-20:
Mine is the solax x3-hybrid inverter, that uses the cloud.

DiedB, 2020-02-20:
@oliviervb So you are sure the local API @Willybits describes does not work?

I could create an implementation that scrapes the Solax Cloud but I doubt whether it will be stable in the long term. Can you send me account details at diederik@outlook.com? I will have a look at implementing it.

Selfie21, 2020-03-11:
Hey I did a really bad and hacky way that scrapes the Solax Cloud. Definetly not recommended to be used, I am still working on it. If used too often it slows down a lot... If you want to try anyways it's the SolaxCrawler: https://github.com/Selfie21/SolaxCloudApi

pieterzen, 2020-05-05:
Hi,
I'm hailing from the Home Assistant community and found this thread. 
I have 1 X1 Boost since yesterday (which uses the Solax Cloud) and can use the following API calls:
http://47.254.152.24:6080/proxy/login/login?password=<pw>&userName=<un> -> Returns token
http://47.254.152.24:6080/proxy/inverter/getDailyInfo?inverterSn=<InverterSn>&wifiSn=<WifiSerial/hostname>&tokenId=<Token from login>&today=2020-01-11
-> Returns all records for given day.

DiedB, 2020-05-06:
@pieterzen Is that your inverter IP address or an API IP address? Does this work for all Solax inverters?

pieterzen, 2020-05-07:
That's the IP of the Solax Cloud server; Currently I'm aware of the following callouts:
http://www.solaxcloud.com:6080/proxy/inverter/getSiteTotalPower?tokenId=<TokenId>&today=2020-05-05&siteId=<SiteId>
http://www.solaxcloud.com:6080/proxy/mysite/getInverterInfo?siteId=<SiteId>&tokenId=<TokenId>

Especially the latter is interesting because it contains the running value's. 

This should work for all inverters that use Solax Cloud. Not Solax Portal. 
More technical details you could look here: 
https://github.com/thomascys/solaxcloud

DiedB, 2020-05-18:
@pieterzen Could you send me your Solax Cloud login credentials (change the password, if you want)? I'll look into implementing it. My e-mail is diederik@outlook.com.

timengaby, 2020-05-18:
Hi Diederik. 

As per message and your request:
Model X1 - 2.0-S-D
Can use solax cloud, did not do that yet.  Also some issues to find my serial number..... Also with the idea to use it with homey.  

[image]

oliviervb, 2020-05-31:
@timengaby I think that diderik is asking the login en pw for the cloud login.

legalr, 2020-06-08:
@DiedB , any progress so far, and did you recieve a Login and PW for testing?

DiedB, 2020-06-09:
Not yet.

locksen-nl, 2020-07-31:
Hey @DiedB  I hope you can get it to work can’t wait 😄 

Any news ?

thijsvgils, 2020-08-12:
@DiedB I'm really looking forward to this too. Can I help in any way? For example sharing login and pw from Solax Cloud?

DiedB, 2020-08-12:
I see there's a lot of interest and I'll see if we can support this for Homey. This thread is quite old already, and no progress has been made for two reasons:

- Confusion about which inverters would be supported
- No known public API available for Solax Cloud

I found [this](https://forum.pvoutput.org/t/solax-cloud-api/3441) thread about a public API. If that works reliably, we can add support for all inverters that support Solax Cloud (which is the majority, I suppose). Could someone try retrieving a token from them? You may also send me your credentials at the email address I posted a few posts back.

thijsvgils, 2020-08-12:
@DiedB email sent! Hope it works:) Here you can find the Solax Cloud API documentation: 
[Solax_API_for_End-user_V1.0.pdf](https://github.com/DiedB/Homey-SolarPanels/files/5063105/Solax_API_for_End-user_V1.0.pdf)

thijsvgils, 2020-08-24:
Any luck yet @DiedB ?

locksen-nl, 2020-08-24:
> Any luck yet @DiedB ?

Can't wait to get it connected to homey

DiedB, 2020-08-25:
Will look into it somewhere in the next few days!

DiedB, 2020-08-27:
Could you try the test version at https://homey.app/a/it.diederik.solar/test/?

The stability of the Solax Cloud API is not great, but it will probably work while sometimes giving an error because of timeouts or connections being refused. As long as it doesn't happen during pairing, you probably won't notice too often.

locksen-nl, 2020-08-27:
> Could you try the test version at https://homey.app/a/it.diederik.solar/test/?
>
>
>
> The stability of the Solax Cloud API is not great, but it will probably work while sometimes giving an error because [...]

The solax cloud is undergoing some server issues atm there upgrading the servers so that could be the stability of the api.

I will test it thnx

oliviervb, 2020-08-27:
Tested and works perfectly,  thx for the support!
I have also some batteries connected to the solax.
Could you please also at the battery status (percentage)

[9 more replies]

closed 2020-09-18 by DiedB