DiedB/Homey-SolarPanels #23: Dummies

kestertonm, 2018-08-29:
I have just downloaded your app and am unable to achieve login to Enphase. Could you please provide some detailed documentation as to how you set this app up?

kestertonm, 2018-08-29:
PS I have API but there is no link back to my solar system that I can ascertain?

DiedB, 2018-08-29:
Create an account at https://developer.enphase.com/. Authorize your API account to access your system (instructions after logging in). Then enter your API key in Homey.

kestertonm, 2018-08-29:
I am unable to find the instructions to authorise my system after logging in

Kind Regards,

*Mike Kesterton *
Phone: 0406-972-510

The information contained in this electronic mail message is privileged and
confidential, and is intended only for use of the addressee. If you are not
the intended recipient, you are hereby notified that any disclosure,
reproduction, distribution or other use of this communication is strictly
prohibited. Confidentiality and legal privilege attached to this
communication are not waived or lost by reason of mistaken delivery to you.
If you have received this communication in error, please notify the sender
by reply transmission and delete the message without copying or disclosing
it.

or mute
the thread

kestertonm, 2018-08-29:
I am not sure what to use as a system ID, I am currently using the serial
number

Kind Regards,

*Mike Kesterton *
Phone: 0406-972-510

The information contained in this electronic mail message is privileged and
confidential, and is intended only for use of the addressee. If you are not
the intended recipient, you are hereby notified that any disclosure,
reproduction, distribution or other use of this communication is strictly
prohibited. Confidentiality and legal privilege attached to this
communication are not waived or lost by reason of mistaken delivery to you.
If you have received this communication in error, please notify the sender
by reply transmission and delete the message without copying or disclosing
it.

On 30 August 2018 at 9:31:28 am, Mike Kesterton (mike@kesterton.net) wrote:

I am unable to find the instructions to authorise my system after logging in

Kind Regards,

*Mike Kesterton *
Phone: 0406-972-510

The information contained in this electronic mail message is privileged and
confidential, and is intended only for use of the addressee. If you are not
the intended recipient, you are hereby notified that any disclosure,
reproduction, distribution or other use of this communication is strictly
prohibited. Confidentiality and legal privilege attached to this
communication are not waived or lost by reason of mistaken delivery to you.
If you have received this communication in error, please notify the sender
by reply transmission and delete the message without copying or disclosing
it.

DiedB, 2018-08-29:
Create an application on https://developer.enphase.com/admin/applications, call it 'Homey' or anything you'd like. Copy the API key you receive (you will need to use it with Homey) and then register your own system by visiting the authorization URL.

kestertonm, 2018-08-29:
Done all of that  …

Got API and User ID

Have no idea when filling in your form in homey what the system is … Have
Tried Serial Number, the number 1, My email ….

All fail login

Kind Regards,

*Mike Kesterton *
Phone: 0406-972-510

The information contained in this electronic mail message is privileged and
confidential, and is intended only for use of the addressee. If you are not
the intended recipient, you are hereby notified that any disclosure,
reproduction, distribution or other use of this communication is strictly
prohibited. Confidentiality and legal privilege attached to this
communication are not waived or lost by reason of mistaken delivery to you.
If you have received this communication in error, please notify the sender
by reply transmission and delete the message without copying or disclosing
it.

DiedB, 2018-08-29:
It's the Site ID, as described on this page: http://www2.enphase.com/enlighten-help/tip/where-do-i-find-my-user-id-and-the-system-site-id/

kestertonm, 2018-08-29:
Yeah I found the same page but this page does not exist in any of the
anaphase tools that I have…..

Kind Regards,

*Mike Kesterton *
Phone: 0406-972-510

The information contained in this electronic mail message is privileged and
confidential, and is intended only for use of the addressee. If you are not
the intended recipient, you are hereby notified that any disclosure,
reproduction, distribution or other use of this communication is strictly
prohibited. Confidentiality and legal privilege attached to this
communication are not waived or lost by reason of mistaken delivery to you.
If you have received this communication in error, please notify the sender
by reply transmission and delete the message without copying or disclosing
it.

http://www2.enphase.com/enlighten-help/tip/where-do-i-find-my-user-id-and-the-system-site-id/

DiedB, 2018-08-29:
Try this guide: https://enphase.com/en-us/support/where-do-i-find-my-system-site-id

kestertonm, 2018-08-29:
I have rung my supplier and they need to contact anaphase to find out ...

Kind Regards,

*Mike Kesterton *
Phone: 0406-972-510

The information contained in this electronic mail message is privileged and
confidential, and is intended only for use of the addressee. If you are not
the intended recipient, you are hereby notified that any disclosure,
reproduction, distribution or other use of this communication is strictly
prohibited. Confidentiality and legal privilege attached to this
communication are not waived or lost by reason of mistaken delivery to you.
If you have received this communication in error, please notify the sender
by reply transmission and delete the message without copying or disclosing
it.

https://enphase.com/en-us/support/where-do-i-find-my-system-site-id

kestertonm, 2018-08-30:
It turns out that the MyEnlighten web app has replaced the instructions and the System ID is no longer visible to the end user.....

DiedB, 2018-11-15:
That's interesting. Did you solve it?

If not, this method of programmatically accessing Enphase inverters might not work anymore, and we will need to look into something new.

closed 2019-03-20 by DiedB