DiedB/Homey-SolarPanels #26: Solarman not connecting anymore

chrisboer, 2018-09-10:
I can connect to http://www.solarmanpv.com/portal/LoginPage.aspx portal 

I don't get the correct data anymore so I tried to reconnect my Solarman again.

When I use the beta app I fill in the username / password and PID but I get after a view minutes the error homey offline and nothing happens.

So I can not re-add my Solarman it happened somewere end of July / begin of August

mchieppa, 2018-09-14:
Same problem here

mchieppa, 2018-10-15:
Is the URL still correct in ..drivers/solarman/device.js or driver.js??

class SolarMAN extends SOTGBase.Device {
    getBaseUrl() {
        return 'http://213.136.73.47:18000/SolarmanApi/serverapi/';
    }
}

API URL for Solarman is: http://www.solarmanpv.com:10000/

Why is there a hard IP address in the code?

DiedB, 2018-11-15:
I directly took over the address used by their mobile app, as the second URL you mention is still hosting their old API.

Does SolarMAN work in the stable version of this app?

mchieppa, 2018-11-15:
Both, stable version and beta version have the same connection problem.
I think because of the fact that http://213.136.73.47:18000 does not exist anymore.
I think Solarman has changed their API url.

The URL I found in the documentation of Solarman is http://www.solarmanpv.com:10000.
This URL leads you to a webservice page
and I think when the old URL is switched with the one above it should be working,
if Solarman did not change their API calls.

mchieppa, 2018-11-15:
I checked the ?WADL, but there are no webservices described in it, maybe we need to look into the updates solarmanpv app to see which URL they use now and adjust the URL to that URL.

DiedB, 2018-11-15:
Can you send your Solarman login details to d@iederik.nl? I'll give the other URL a try, but I do not think it's going to work. The app uses a different API now, which is currently incompatible with this app.

mchieppa, 2018-11-15:
I don't think it will work, because the URL has no API calls

I see the app is now connecting to http://47.91.240.139:18000/ [Hong Kong],
but I don't get any response when I try it in a browser

mchieppa, 2018-11-15:
I'm also getting a second URL in the capture: 203.205.146.45:80,
this is going to a nginx webserver

chrisboer, 2018-11-18:
http://www.solarmanpv.com:10000 is the same as http://47.91.240.139:10000/ I will try it with an updated IP address

chrisboer, 2018-11-18:
Yes it works with http://www.solarmanpv.com:18000/SolarmanApi/serverapi/
Just changed it and connected Solarman again, I left my original also in so it should now both work with 
Zonnepanelen v2.1.0 Diederik Bakker Geïnstalleerd via Developer Tools

mchieppa, 2018-11-21:
It's working only when you use site id, NOT site name!

chrisboer, 2018-11-21:
> 
> 
> It's working only when you use site id, NOT site name!

What do you mean with site id because it was not working for me, yesterday I found out that it does connect the Solarman but not giving the values tonight I will put in the other fork and will recheck if I can add the device but I think this issue or the other issue are the same so one of them should be closed to keep all the info in one place

mchieppa, 2018-11-21:
I agree that all information should be on one place, that's why I added the Solarman extra information in this closed thread. When configuring Solarman Solarpanels you fill in 3 things, username, password and sitename or siteid. Sitename is not working and will time out, site id is working

DiedB, 2018-11-21:
I lost track, is this about SAJ or Solarman?

DiedB, 2018-11-22:
@chrisboer Does Solarman work on 2.3.2 alpha? Please confirm so I can publish it in the app store.

mchieppa, 2018-11-22:
I used the alpha 2.3.2 version and I can make a connection to Solarman
and today I checked that the energy parameters are filled with correct values

chrisboer, 2018-11-22:
I don’t get a connection with Solarman it keeps trying to connect.
I tried to connect directly to the website login that was working so the site was not offline.
But if mchieppa tested it ok it could be I used a wrong version. So push it @diedb

DiedB, 2018-11-22:
I have updated the app store beta to 2.3.2. Can you try installing and testing it?

chrisboer, 2018-11-22:
Yes @DiedB it works ... I can add it now and I see already the energy total of today.

closed 2018-11-22 by chrisboer

aamirmalek, 2020-05-28:
I am working on the project where I want https://www.solarmanpv.com's data into MTSQL using the API calls to the website. Any help?

DiedB, 2020-05-28:
What do you mean? You can look at the code and use it under the GPLv3 license.