DiedB/Homey-SolarPanels #27: SAJ not supported

chrisboer, 2018-09-11:
I installed the BETA version but I see SAJ is not supported there anymore why ?

DiedB, 2018-11-15:
Does SAJ still work with the main version? I am slowly porting over inverters to the new version, but SAJ has not been ported yet. This will be done before Athom removes SDK1 support.

DiedB, 2018-11-18:
Which inverter model do you have? Where do you monitor it currently?

chrisboer, 2018-11-18:
powerone / solarman 

http://www.solarmanpv.com/portal/LoginPage.aspx

I monitor it here after I type in my userid and password

chrisboer, 2018-11-18:
It has been resolved by changing 

'use strict';

const Homey = require('homey');
const SOTGBase = require('../base');

class SolarMAN extends SOTGBase.Driver {
    getBaseUrl() {
        return 'http://www.solarmanpv.com:18000/SolarmanApi/serverapi/';
    }
}

module.exports = SolarMAN;

DiedB, 2018-11-18:
I'll fix this in the app and release it to the app store, thanks a lot for investigating!

chrisboer, 2018-11-19:
I looked at it today but I don't see the Solarman changed ? So still no data is retrieved ?

DiedB, 2018-11-20:
Can you try the app version on the alpha branch of this repository? It should work there, let me know if it doesn't!

chrisboer, 2018-11-21:
> 
> 
> Can you try the app version on the alpha branch of this repository? It should work there, let me know if it doesn't!

I will try this tonight to see if I can download this part but I sometimes have issues going to another branch... keep you posted with the results

mchieppa, 2018-11-21:
You can do a: git clone -b alpha https://github.com/DiedB/SolarPanels.git
Then you will force it to download the alpha branche

chrisboer, 2018-11-21:
Thanks @mchieppa I will try after dinner, learning all the time ;-)

chrisboer, 2018-11-21:
Got the code installed but I can not add a Solarman so it does not connect nor gives an error... also I also noticed that SAJ inverters are missing in this release ?

DiedB, 2018-11-21:
I lost track, is this about SAJ or Solarman? Solarman should have been fixed in 2.3.2 (which is not in the Athom app store yet), SAJ has never been ported to the new version.

chrisboer, 2018-11-21:
@DiedB Solarman is not working anymore, SAJ is not in the current BETA version available, SAJ is available and working in the v1.4.3 - 12.06.17

[image]

mchieppa, 2018-11-22:
This thread is about SAJ, please only discuss SAJ here. So there is no confusion.
Solarman is in the other thread:
https://github.com/DiedB/SolarPanels/issues/26
The Solarman thread is closed because it is working in alpha version 2.3.2.

DiedB, 2018-11-22:
Okay. Does SAJ work for everyone, in the implementation that is currently in 1.4.3? If so, I will port it over to 2.4 soon.

chrisboer, 2018-11-23:
> Okay. Does SAJ work for everyone, in the implementation that is currently in 1.4.3? If so, I will port it over to 2.4 soon.

Yes that works perfectly so can be ported over to 2.4 ... thanks !

chrisboer, 2019-01-06:
> 
> 
> Okay. Does SAJ work for everyone, in the implementation that is currently in 1.4.3? If so, I will port it over to 2.4 soon.

Hi @DiedB when do you think you will be able to port over the SAJ part into the BETA or final new release as SAJ is in latest normal release but Solarman is only available in beta app and in the beta app SAJ is not available.

Hope to hear from you soon... 

by the way happy new year all

DiedB, 2019-01-06:
I'm very busy with my thesis, so I will probably not have time until the beginning of February. After that, I'm happy to implement it.

A-Industry, 2019-02-22:
Hi, I would love to see SAJ ported to version 2.x as well. No pressure :)

speedyco, 2019-02-24:
Hi Diederik, In Homey 2.0.2 and app 1.4.3, NULL values are stored from the SAJ Inverter. I'd love to see SAJ ported as well. Good luck with your thesis!

chrisboer, 2019-03-13:
> I'm very busy with my thesis, so I will probably not have time until the beginning of February. After that, I'm happy to implement it.

Hi Diederik you are probably celebrating your goals... but how can I help you to merge this together ?

DiedB, 2019-03-20:
Just added Fronius back, SAJ will be next! Sorry for the long delay.

chrisboer, 2019-03-20:
Thanks Diederik ... no problem you had other priorities which are more important than looking at solarpanel supply 😁

chrisboer, 2019-04-01:
Hi Diederik any view on SAJ being added the sun is shining more often now so I wpuld love to know my daily profits ;-)

DiedB, 2019-04-03:
Still on my todo list, hopefully this week!

DiedB, 2019-04-03:
It will however be paired with some other substantial changes and released as 3.0, so it might take some more work than just the porting over of SAJ inverters.

speedyco, 2019-04-03:
> It will however be paired with some other substantial changes and released as 3.0, so it might take some more work than just the porting over of SAJ inverters.

We appreciate the effort and the work you do!

DiedB, 2019-04-12:
I have re-added SAJ support and it will be released in 3.0.0 beta. Should be live within 1-2 days from now. I do not have anything to test this with, so please try it and report back!

chrisboer, 2019-04-12:
Thanks Diederik I will test it right away today as I will install it in developer mode

chrisboer, 2019-04-12:
Hi Diederik, 
I have installed the github beta 3.0.0

When adding SAJ the device homey stopped... 

When looking in solarpanels I see SAJ is added but I get this error.
it is not working I get Error retrieving data (ReferenceError:_ is not defined)

Access should be fine as I can open them in my browser

[image]

Please let me know if I can help testing this with you, I can also port forward access to my solarsystem from internet so you can check it yourself / test it.

DiedB, 2019-04-12:
That would be great. Can you add NAT rules and send me the IP address and port on diederik@outlook.com?

DiedB, 2019-04-12:
Whoops, there was a programming error. I have fixed it, can you test the current GitHub version @chrisboer?

chrisboer, 2019-04-12:
Same issues I have also same beta version so did you commit

DiedB, 2019-04-12:
My bad. Should have been pushed now.

chrisboer, 2019-04-12:
I will check it later... I now have to go ... will check it later or tomorrow morning

chrisboer, 2019-04-12:
> 
> 
> My bad. Should have been pushed now.

Sorry I don't want to be rude but SolarMan is now disappeared again. So you add SAJ but removed SolarMan in the beta. !
I thought you would integrate SAJ in ... as SolarMan was already in this should not disappear.

Please let me know the next steps so I can check a new beta

chrisboer, 2019-04-13:
add the first SAJ with adding the second I got the error duplicate_device but it has a different IP

DiedB, 2019-04-13:
Thanks for the feedback. I thought the Homey SDK would just generate IDs in the case that there is no unique differentiator between devices, but apparently it does not. I have added this and you should now be able to add multiple SAJ systems.

SolarMan will be back soon, it is called beta for a reason.

chrisboer, 2019-04-13:
but the one added works perfect

DiedB, 2019-04-13:
Can you try SolarMAN in the newest commit on the beta branch?

chrisboer, 2019-04-13:
Yes they all work, also tested SAJ / Solarman and multiple devices

[image]

Thanks remove the beta and put in it in production ;-)

closed 2019-04-19 by DiedB