DiedB/Homey-SolarPanels #29: Lost connection with ZeverCloud

hplos, 2018-10-31:
In my homey, my solar panels are broken. cause:

error retrieving data (FetchError: invalid json response body at https://www.zevercloud......... reason: Unexpected token < in JSON at position1)

DiedB, 2018-11-15:
Are you still experiencing this issue?

hplos, 2018-11-15:
> Kun je mij je logingegevens sturen op d@iederik.nl? Eventueel tijdelijk
> even je wachtwoord veranderen. Ik zal even kijken wat er aan de hand is.

Ik wil niet vervelend zijn, maar dat ga ik echt niet doen.

K wil best even je project colonen van git en een debug versie draaien en
je logs sturen, maar ik ga je echt geen login geven van mijn homey

Sorry, no can’t do

Graag hoor ik van je!

Vriendelijke groet,
Hans-Peter Los

notifications@github.com>

DiedB, 2018-11-15:
Het gaat mij niet om Homey maar om Zevercloud. Voor mij helaas de enige manier om snel dit soort issues te troubleshooten, ik heb die omvormers niet zelf.

Tenzij je genoeg kunde hebt om de Zevercloud-API zelf aan te spreken op de manier die mijn app gebruikt, dat mag natuurlijk ook. In dat geval kun je me de request sturen en de gegevens censureren.

hplos, 2018-11-15:
In dat geval denk ik dat je genoeg hebt aan de api key van mijn solar login?

Dat is het enige wat ik in jouw app invoer.

a4f295679bd340698b3785ea89147c71

DiedB, 2018-11-15:
Lijkt erop dat hun hele API niet meer werkt. Er is wel een v2, maar ik heb geen documentatie.

Ik vrees dat er voorlopig geen manier is om Zeversolar uit te lezen vanuit Homey. Kun je er een aparte kWh-meter tussen zetten en de data naar een ander platform pushen, zoals PVOutput? Dat is een alternatieve manier om de link met Homey te leggen. Je kunt ook een losse kWh-meter nemen die direct met Homey verbindt.

DiedB, 2018-11-15:
Voor PVOutput moet je zelf data pushen, en daar heb je extra hardware voor nodig. Er zijn een aantal uitzonderingen waarbij PVOutput de data zelf op kan halen bij de leverancier van je omvormer, maar dat gaat in dit geval niet op (of zij hebben een API van Zeversolar die ik niet ken).

Ik heb een keer een setup gemaakt met een YouLess v2 en een slimme meter, maar die heb ik inderdaad verbonden met zowel een aparte kWh-meter voor de zonnepanelen en de slimme meter zelf. YouLess zal op basis van alleen je elektriciteitsmeter nooit het terugleveren kunnen meten.

hplos, 2018-11-15:
Denk dat de Youless, wanneer die aan is gelopen op een slimme meter heel goed de teruglevering kan meten, maar niet wat je panelen opbrengen.

Waar ik beetje mee bezig ben is dat ik graag mijn energie goedkoop op wil slaan zodra in terug ga leveren. Nu nog niet interessant, want mijn oude domme meten saldeert heel netjes voor me, maar deze zal ooit een keer vervangen gaan worden denk ik.

Ben nu aan het onderzoeken of het handig is om energie in warm water op te slaan zodat ik (helemaal in de zomer) kan besparen op mijn gas.

Dat terzijde.

Bedankt voor de moeite en succes met je app. Ik hoop dat m’n zevercloud het binnenkort weer gaat doen met je appje.

Ben ook erg benieuwd naar homey 2.0 :D

Groetjes!

DiedB, 2019-04-19:
Closing this issue because there is currently no (documented) API. Please report back if you find anything.

closed 2019-04-19 by DiedB