DiedB/Homey-SolarPanels #31: Flow error

HermanHSlagman, 2018-11-28:
Hi Diderik,

I have added and configured my SolarEdge PV system in my Homey with **your** app. Test with this configuration works fine.
Now I will create a workflow to activate a power consuming device if the power is higher than a specified level.
I use the "Logica" app to specify the power level. Similar to your example (_I think_).
But I get an unexpected error message when I test my flow. 
In your example, I see the Power label; I can't select this Power label.
My test flow:
 [image]

Detail of the flow:
[image]

Error message:
[image]

I'm not a technician, but can you help me? Is it possible for you to give me some hints or give me more detail examples?

Thank you in advance.

DiedB, 2018-11-28:
There's small arrows right of the leftmost card, if you hover it. Click one of them and you get a power card. The blue tag can then be dragged into your middle card.

closed 2018-11-28 by DiedB

HermanHSlagman, 2018-11-29:
Thank you for your hint, Diederik,

It’s easy if you know the solution.

Regards,
Herman.