DiedB/Homey-SolarPanels #34: Fronius support in beta app?

gjefsjo, 2018-12-23:
I have used Solar Panels app with Fronius and was very happy with it. 
Now when using v2.0 i can't use the 1.4.3 app that supports fronius. 
Can you add Fronius support in beta app?

DiedB, 2018-12-24:
I heard it was possible to still run SDK1 apps (the stable version of this app) on Homey v2.0 but I might be wrong. I will migrate all inverters brands that aren't already in the new version as soon as possible.

gjefsjo, 2018-12-24:
I get the "this app is incompatible with your homey software version" message[image]

DiedB, 2018-12-24:
I see, it seems that users that had the app installed before upgrading to 2.0 still are able to use it, but new installs are not possible.

gjefsjo, 2018-12-25:
yes, that's right. but i did not get any readings from the inverter. so i uninstalled in the belief that should be using the beta. i am very happy you have made this app, it solves a big problem for me. i live on a offgrid farm and this helps me utilize the solar energy better, only have the heat on when the sun is shining is an example. Thank you for doing this!

jigern, 2019-01-17:
I hope you can fix to fronius . i have now homey 2,0 thanks

andrewbadge, 2019-01-22:
Hi, I'm new to Homey and v2.0. I can't see Fronius as a choice when adding a device so I assume this is related to this issue. +1 for Fronius support in the beta please.
Thanks for your work.

Zork9999, 2019-01-24:
Yes here also an upvote for supporting Fronius in v2.0

Abarth695, 2019-01-25:
Hi
Fronius no longer working or supported
Why?
Can this please be fixed

DiedB, 2019-01-25:
Will add support back next week.

pln8421, 2019-02-10:
Hello
Fronius is no 2 after SMA and should get higher priority than Medium :)
I hope ”next week” is near. 
Best regards, Per in Sweden

gjefsjo, 2019-03-04:
The sun is getting stronger everyday now! how is development going? will you soon have an app for us?

andrewbadge, 2019-03-05:
> The sun is getting stronger everyday now!

Opposite for me (in the southern hemisphere)..but still keen to get this working!

Abarth695, 2019-03-05:
Is Fronius fixed yet?

Sent from Angelo’s  iPhone Xs

pln8421, 2019-03-19:
Hi. Can you explain why nothing happens? Do you have something against Fronius because you add so many other much smaller brands. 
Can someone else do it?

DiedB, 2019-03-19:
Sorry for the long delay. Fronius support has been added back in the newest beta version which is under review by Athom. Should be published in a few days.

Because I have nothing to test this with, I cannot yet confirm whether the new code works (it's a straight migration of the old Fronius code). Please try it and report back.

DiedB, 2019-03-20:
The update has been published. Please test it and report back whether all is working.

gjefsjo, 2019-03-21:
I have tested the app now, at the end of the setup it made the homey app crash, but it installed the inverter so i’m Happy!

pln8421, 2019-03-21:
Hello
It did not work well. I got update.
First time homey app chrashed when trying. Restarted app and tried again ang it went almost through. Got a device with triangle. Did not work. Removed. Restarted app and Homey.
When trying i get error in the end. See screen shots below.
First time
[Image]

Then again
[Image]

[Image]

[Image]

When click in IP field I get this

[Image]

When starting writing 19

[Image]

[Image]

Router. I use the right IP
[Image]

And it work

[Image]

Best regards Per

Abarth695, 2019-03-21:
Working on my Fronius.
Thanks 

Sent from Angelo’s  iPhone Xs

pln8421, 2019-03-21:
Very good :)
If you want I can later do port forwarding on port 80 to my fronius and give Wan IP.

Best regards Per

Hälsningar Per

Abarth695, 2019-03-21:
Stopped working
Worked for a while then this
Anything I can check
Thanks

Sent from Angelo’s  iPhone Xs

DiedB, 2019-03-21:
Can you please post the relevant images directly on GitHub? When you put images in the email, I cannot see them unfortunately.

pln8421, 2019-03-21:
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]

DiedB, 2019-03-21:
@pln8421 Does the device become available again? EHOSTUNREACH implies that the inverter cannot be reached on that IP address.

I cannot reproduce the black box you get while typing. What type of iPhone is this? I have an iPhone XR.

The error handling during the pairing process seems to be broken for all inverters somewhere since Homey v2.0 and after a lot of digging, it seems like it is something on Athom's part. This causes the app to crash after pairing, even though the device is added succesfully.

I will move towards 3.0 while I implement some other ideas for the app and in that version, everything will be fixed.

pln8421, 2019-03-21:
Hello.
I have Iphone 6. I tried to add again and figured out the first time I press connect it take time. Therfore I pressed connect again before
Yes the app crash and you are right. The device is added and Work :). I can see both kW and kWh now.
The instruction window, perhaps the background colour is transperent? because the background is black
How do you act when fronius data manager goes into sleep mode to save energy? Their app say offline. I would like to see 0kW. What I think I remember from last year there was a frozen value bigger than 0.

Best regards Per

DiedB, 2019-03-21:
> How do you act when fronius data manager goes into sleep mode to save energy? Their app say offline. I would like to see 0kW. What I think I remember from last year there was a frozen value bigger than 0.

Good question, and I do not know for sure. It depends on what the API does when the Datamanager is sleeping. Can you check the app's behaviour in that situation and report back?

pln8421, 2019-03-21:
Yes I will do that. Thanks

Hälsningar Per

pln8421, 2019-03-21:
My earlier question:
How do you act when fronius data manager goes into sleep mode to save energy? Their app say 16W, stay then in 0W and goes into sleep-mode offline. Problem is the Tag freeze with a value the whole night when this happens.

My investigation
Low production: 
[image]

0W
[image]

After 0W is reached your APP still show 29W
[image]

Datamanager goes into sleep mode and Device is unavailable from Homey:
[image]

But Tags still show 29W
[image]

The default setting is datamanager goes into sleep mode to save energy. I think this setting can be changed so it keeps the communication up and running.

Best Regards, Per

DiedB, 2019-03-21:
Can you go to http://192.168.0.67/solar_api/v1/GetInverterRealtimeData.cgi?Scope=Device&DeviceID=1&DataCollection=CommonInverterData and post a screenshot of the data you see?

pln8421, 2019-03-21:
{
   "Body" : {
      "Data" : {
         "DAY_ENERGY" : {
            "Unit" : "Wh",
            "Value" : 5902
         },
         "DeviceStatus" : {
            "ErrorCode" : 523,
            "LEDColor" : 3,
            "LEDState" : 0,
            "MgmtTimerRemainingTime" : -1,
            "StateToReset" : true,
            "StatusCode" : 2
         },
         "TOTAL_ENERGY" : {
            "Unit" : "Wh",
            "Value" : 8058579.5
         },
         "YEAR_ENERGY" : {
            "Unit" : "Wh",
            "Value" : 844656.5
         }
      }
   },
   "Head" : {
      "RequestArguments" : {
         "DataCollection" : "CommonInverterData",
         "DeviceClass" : "Inverter",
         "DeviceId" : "1",
         "Scope" : "Device"
      },
      "Status" : {
         "Code" : 0,
         "Reason" : "",
         "UserMessage" : ""
      },
      "Timestamp" : "2019-03-21T19:12:03+01:00"
   }
}

DiedB, 2019-03-21:
Thanks! Can you do the same tomorrow, while the inverter is not in standby? I want to see if any status codes change, that would help in programming the app's behaviour.

pln8421, 2019-03-21:
Ok

Hälsningar Per

pln8421, 2019-03-22:
Up and running:
{
   "Body" : {
      "Data" : {
         "DAY_ENERGY" : {
            "Unit" : "Wh",
            "Value" : 161.69999999999999
         },
         "DeviceStatus" : {
            "ErrorCode" : 0,
            "LEDColor" : 2,
            "LEDState" : 0,
            "MgmtTimerRemainingTime" : -1,
            "StateToReset" : false,
            "StatusCode" : 7
         },
         "FAC" : {
            "Unit" : "Hz",
            "Value" : 49.969999999999999
         },
         "IAC" : {
            "Unit" : "A",
            "Value" : 2.0600000000000001
         },
         "IDC" : {
            "Unit" : "A",
            "Value" : 3.3700000000000001
         },
         "PAC" : {
            "Unit" : "W",
            "Value" : 465
         },
         "TOTAL_ENERGY" : {
            "Unit" : "Wh",
            "Value" : 8058739.5
         },
         "UAC" : {
            "Unit" : "V",
            "Value" : 229.80000000000001
         },
         "UDC" : {
            "Unit" : "V",
            "Value" : 150.59999999999999
         },
         "YEAR_ENERGY" : {
            "Unit" : "Wh",
            "Value" : 844818.18999999994
         }
      }
   },
   "Head" : {
      "RequestArguments" : {
         "DataCollection" : "CommonInverterData",
         "DeviceClass" : "Inverter",
         "DeviceId" : "1",
         "Scope" : "Device"
      },
      "Status" : {
         "Code" : 0,
         "Reason" : "",
         "UserMessage" : ""
      },
      "Timestamp" : "2019-03-22T06:41:24+01:00"
   }
}

DiedB, 2019-04-12:
I have made the improvements (standby power value, pairing screen bugs, pairing error handling), and they will be released in 3.0.0 beta. Should be live within 1-2 days from now.

This update will require you to remove and re-add your system.

gjefsjo, 2019-04-12:
Adding my system made the homey app cash, but the system are added😃

DiedB, 2019-04-13:
The Homey app crashing on adding the system is expected, will be fixed in the next beta.

I'm mainly wondering if the data correctly shows 0W at night and whether the inverter does not go offline in the Homey app at night.

gjefsjo, 2019-04-13:
I have removed and added several times bit i get an errormessage : Currentpower is not defined.

DiedB, 2019-04-13:
Whoops, my fault. Will be fixed in the next beta. Until then, feel free to use the stable channel.

DiedB, 2019-04-13:
v3.0.2 will be released in a few days (when Athom approves). This will require you to remove and re-add your Fronius inverters, after that everything should work smoothly!

closed 2019-04-13 by DiedB