DiedB/Homey-SolarPanels #35: Adding Goodwe device not working

martinhilgenkamp, 2019-01-16:
I lost connection to the Goodwe portal and tried to reconnect.
If i select Goodwe in the add device panel and start typing my username the app scrolls down and shows a black box. After this event the connect button is not functioning anymore.

Some info:
Platform: Apple iPhone 7 iOS 12.1.3
Homey: 2.0.0
Zonnepanelen: 2.4.0 (beta)

DiedB, 2019-01-17:
Can you try it on https://developer.athom.com/tools/pair?

Markmenh, 2019-01-18:
I have the same problem.
And also via the link didn't work.

Homey: 2.0.0
Zonnepanelen: 2.4.0 (beta)

DiedB, 2019-01-18:
Does the Goodwe Portal work? And/or their app?

meden199, 2019-01-19:
Goodwe is changing the portal.
Goodwe portal wil be closed. SEMS portal is the replacemant.
https://www.semsportal.com

DiedB, 2019-01-19:
Can you send me your username? I'd like to have a look at what's going wrong here.

meden199, 2019-01-19:
My username is meden199

Verzonden vanaf mijn Samsung Galaxy-smartphone.

bart973, 2019-01-19:
Volgens mij: Smeck_solar

Groeten 

Bart Meppelder

DiedB, 2019-01-19:
I see, the problem is probably with this app. I will look into it after the weekend.

bart973, 2019-01-19:
Tanks👌👌

someren76, 2019-01-28:
I have just installeren 2.4.0 on a new homey. When I enter my username I cannot press continue. I do not know if this is the same problem. But just wanted to let you know.

martinhilgenkamp, 2019-01-28:
Hi sorry did not send my answer. On the Developer Website i have the same problem.

Just saw an update from another user that hè has the same problem aswell.

With kind regards
Martin

meden199, 2019-02-10:
Gooedwe Portal is down since 30 october 2018.

Login is disabled and we have to use SEMS portal. (semsportal.com)

If the android app is still working I don't know, I also switch to the SEMS app

macfreek, 2019-02-22:
+1. Would be great if SEMS portal is supported, as the old GoodWe portal is closed for a few months now.

iformarco, 2019-03-20:
Also hoping to monitor my Goodwe transformer into the Homey...

DiedB, 2019-03-21:
See #28.

closed 2019-03-21 by DiedB