DiedB/Homey-SolarPanels #36: Pvoutput not updating

kevvie, 2019-01-16:
I've updated my homey to v2.0 and running the solar panels beta app(v2.4.0) but pvoutput is syncing for a few minutes and stops randomly.

DiedB, 2019-01-16:
Sometimes the cron jobs break when upgrading to 2.0. Can you remove the PVOutput device, then the app entirely and then reinstall the Solar Panels app (beta) and add your PVOutput device again?

kevvie, 2019-01-18:
I've tried it but the problem still exists.

I've added a screenshot from insights. I own 2 inverters A and B and I have one pvoutput system which contains them both combined. The pvoutput website does show the "live" data but it isn't available in the homey app for two of my systems.
[image](https://drive.google.com/file/d/1z2Jij3ey8fTONi4Fnf9p6hL6McQW5jNs/view?usp=drivesdk)

DiedB, 2019-01-19:
What do you mean by syncing and stopping? Insights collects the data just fine, my PVOutput system works fine in the app as well. How do you check the energy and power in the app?

kevvie, 2019-01-20:
Ive done some more testing today. It seems like adding another pvoutput system breaks the previous added systems.

When I have only 1 system connected there is no problem, the data is updating like it should. When I add another system, the first system stops updating.

DiedB, 2019-01-20:
Thanks for the clarification, I will look into it.

DiedB, 2019-02-08:
Although I do not have multiple systems to test it with, you might be hitting API rate limits. Are both systems under the same account, while having different system IDs?

This app will fire 12 requests per hour per system, which will add up to 24 for two systems under the same account. If the service that is pushing the data to PVOutput is using more than 48 requests per hour, your API key will be rate limited until the next hour.

Could you try upgrading your PVOutput account for a year and check whether this solves the issue?

LMSFan72, 2019-02-26:
This looks similar to my issue that I created separately on the solarman. Mine doesn't appear to be API limits since one of them still updates. One thought I have is to create a "second" app and install that as well and see if one inverter can be linked to one, and the second to another. Unfortunately, I don't know how to do that but I envisage something like "SolarPanels2ndInverter" being the second app. In that case I would have "SolarPanels" and "SolarPanels2ndInverter" installed on Homey....

DiedB, 2019-03-06:
Homey v2.0.5-rc.3 Experimental has a line in the changelog that might fix this issue:

> [Apps SDK] Fixed a bug where CronTasks would fire the wrong event

Could you please install that version and verify whether this fixes your issue? If it doesn't, please try to remove the inverters and the app entirely and recreate them again. Please report back!

closed 2019-04-02 by DiedB