DiedB/Homey-SolarPanels #37: Add consumption from grid

solpojke, 2019-01-17:
A feature to consider and which would be very appreciated is monitoring of overall consumption from the grid.

This feature is available in e.g. Solaredge. 

It will create possibilities to turn off peak loads etc

DiedB, 2019-01-17:
I'd like to add support for this to the app. How did you get grid monitoring to work with your SolarEdge inverter? Did you buy another network-connected appliance from SolarEdge that measures at your net meter?

DiedB, 2019-01-17:
I see. Can I borrow your API key to do some testing? You can send it via e-mail to d@iederik.nl.

DiedB, 2019-03-20:
Working on it but got stuck on an issue on Athom's side. Will wait for their response before proceeding.

https://github.com/athombv/homey-apps-sdk-issues/issues/76

DiedB, 2019-03-21:
I have implemented the app with the `powerDetails` and `energyDetails` endpoints. However, because   more (and different) capabilities are needed to pass the data to the Homey SDK, all existing SolarEdge installations would break immediately after updating the app and SolarEdge systems would have to be re-added manually.

I think I am going to release this feature in 3.0 on the beta branch (and add other inverters with consumption data as well) and slowly wait until users adopt it. I'll keep you up to date.

DiedB, 2019-04-13:
Support has been added in v3.0.2, should be in the app store in a few days.

closed 2019-04-13 by DiedB