DiedB/Homey-SolarPanels #39: Energie reporting incorrect value for Enphase systems

RobKoenis, 2019-02-04:
I have linked my Enphase system to homey. The connection is working en it is reporting values. The Energiemeter value seems to be correct. But the reported Energie value is not.
I have checked the enphase api. It seems the summary api you use is not reporting correct values for current_power.
This is not only on my system. Also other system have the same problem. 
See:
https://developer.enphase.com/forum/topics/summary-current_power-reliable

As replacement for the summery api you could use the stats api. This seems to reporting correct values.
This is reporting the power in intervals of 5 minutes.
Without any filtering it is reporting all intervals for the current day (adding a filter can make the result smaller and faster to process).
You can then take the last interval. 
If the last interval is too old (system is not reporting intervals when system is not generating power) then 0 can be used for the energie value.
This also solves the problem that current_power never gets to 0 in the summery api.

Sample of a stats call:
https://api.enphaseenergy.com/api/v2/systems/1234567/stats?key=<key>&user_id=<userid>&datetime_format=iso8601

Sample output (most intervals removed):

{
system_id: 1234567,
total_devices: 8,
intervals: [
{
end_at: "2019-02-04T08:00:00+01:00",
devices_reporting: 2,
powr: 0,
enwh: 0
},
{
end_at: "2019-02-04T08:05:00+01:00",
devices_reporting: 8,
powr: 2,
enwh: 0
},
.
.
.
{
end_at: "2019-02-04T17:00:00+01:00",
devices_reporting: 8,
powr: 37,
enwh: 3
},
{
end_at: "2019-02-04T17:05:00+01:00",
devices_reporting: 8,
powr: 33,
enwh: 3
}
],
meta: {
status: "normal",
last_report_at: "2019-02-04T17:52:52+01:00",
last_energy_at: "2019-02-04T17:05:41+01:00",
operational_at: "2019-01-31T13:18:16+01:00"
}
}

DiedB, 2019-02-08:
Thanks for the detailed report, I will look into it.

DiedB, 2019-03-20:
I do not really have a system to test this with. Can you confirm that always getting the last index from the stats array, checking if the end_at is different from the latest value that was reported to Homey and if it is, adding powr as new value for currentPower?

RobKoenis, 2019-03-20:
Yes, the last item is having the last reported power.
At the end of the day the last value reported is 0. 
When the system is not generating any power its stopping the 5 minute intervals.
I can send you a account to check it on a life system (my system).

RobKoenis, 2019-03-20:
I have send you account info using facebook (did not see a option to send a private message in github)

DiedB, 2019-03-20:
I assume that `powr` needs to be multiplied by `total_devices` to get the total average power?

Is the daily production report correct using the summary API?

DiedB, 2019-04-13:
Hi, can you please report back with answers on those two questions? I'd like to get this fixed in the current beta release.

RobKoenis, 2019-04-15:
No, the powr in the stats api is of all the devices, No mulitiplie needed.

DiedB, 2019-04-15:
Can you send me your e-mail address? I will add you to an alpha version so we can test.

RobKoenis, 2019-04-15:
Email adres send (facebook message)

Robar56, 2019-04-15:
Bij deze ;-)

DiedB, 2019-04-15:
@Robar56 That doesn't work, GitHub strips your email address when you reply by email.

@RobKoenis I have added you to alpha v3.0.3, please try it out! You have to remove your system and the Solar Panels app first, after that you can install and add everything again. In the current implementation, only the stats endpoint is being used. I'm summing energy (kWh) values, can you double check whether those values are correct?

RobKoenis, 2019-04-15:
The 3.0 that was auto updated 2 days ago did not seem to work (401 error).

The 3.03 alpha installed. Nice that systemid was no longer needed (lots of user don't know that one). 
Do jou also support if one account has more than one system (i hava only one)?

After installing the values where empy.
After some time the values where filled with the correct values.
I will follow it to check if it will update with the correct values.

DiedB, 2019-04-15:
3.0 is broken and was accidentally pushed to the stable channel (whoops).

It should support multiple systems just fine, but I do not have any Enphase users to test that with.

Values should be empty for a little while after pairing, because it goes in 5-minute intervals. By the way, that is a good one to improve across all inverters and I will do that soon.

Please test it for a few days and check whether everything works as expected. Also, check whether the total daily energy value matches that of the Enphase app.

RobKoenis, 2019-04-16:
The values are correct. The production was about 50..100 Wh behind the value in the Enphase app or website when the system was at full power/production. But at the end of the day when the production was low they did have the about same values. 

But after the energy production was stopt (all micro converters are down and no new data is send to enphase ) the homey app keeps showing the last production (20W yesterday). If you click on the value you can see how many minutes ago the last update was done. This time is going up far beyond 5 minutes but the value shown stays 20W. This must go to 0W if no new intervals are added after some time. 

When clicking on the values in the homey app you can switch the description to the time the last update was done. But this time is not automatically updated. Solution is to switch back and forth between description and time does a update. But would be nice if it is updated automatically. 

But then after midnight, a new day, and no sun yet. 
Then the homey app did show:
Apparaat niet beschikbaar
Error retrieving data (TypeError: Cannot read property 'join' of undefined)

I checked the stats api, it did return status code 422 with the follow data:
{
reason: "Requested date range is invalid for this system",
start_at: "2019-04-16T00:00:00+02:00",
end_at: "2019-04-15T20:25:00+02:00",
first_interval: "2019-01-31T13:18:16+01:00",
last_interval: "2019-04-15T20:27:58+02:00"
}
If no datetime format is specified it returned:
{
reason: "Requested date range is invalid for this system",
start_at: 1555365600,
end_at: 1555352700,
first_interval: 1548937096,
last_interval: 1555352878
}
I did not specify any date/time filter. The default would be the the range for this day but there is no data yet for this day. 

When the system did start producing energy this morning the app did NOT start working correct. 
Even after I restarted the app the error did stay the same: Error retrieving data (TypeError: Cannot read property 'join' of undefined). The stats api does report normal info at this time. Seems the internal state of the app is broken.

DiedB, 2019-04-16:
Thanks a lot for the extensive feedback! I will look into it and push a new alpha version in a few days.

RobKoenis, 2019-04-17:
Additional info,
In the homey app the device looks broken (red icon and error on device screen: TypeError: Cannot read property 'join' of undefined) after the first day. But in the background it was still updating its values. In homey insights I see the value was updated the last 3 days.

DiedB, 2019-04-19:
Please try v3.0.6 and try if that solves the issues.

RobKoenis, 2019-04-19:
After the upgrade (auto) i did still have the error in the homey app: Error retrieving data (TypeError: Cannot read property 'join' of undefined).

I removed the Enphase device in homey and installed it again. The error was gone. But the value are 0 and did stay zero.

Then i completly uninstalled the app and installed it again and created the device again.
Values did stil stay 0

DiedB, 2019-04-19:
I think you solved the issue already by removing and re-adding (I suspect
an SDK bug keeping the device unavailable after auto-upgrade). Let me know
if it ever stays unavailable again.

RobKoenis, 2019-04-19:
I did not solve the issue of the 0 value. This is a new problem. Both values stay zero, also after reinstall. The value never gets updated.

DiedB, 2019-04-20:
Apparently the date range parameters in the Enphase API are not working as I expected. I pushed another fix on the alpha branch, please try and see if it works for you.

RobKoenis, 2019-04-20:
Now running 3.0.7 alpha. Does seem to work ok. 
Will check if it still works when there is no sun and after a day switch.

RobKoenis, 2019-04-21:
Second day, sun is shining and it reports correct values.

But at the end of yesterday when the sun was down it did stay at 4 watt while the system was not generating any more. 
This morning before the sun was up and the system was not generating anything the device in the homey app did show the error:
Apperaat niet beschikbaar
Error retrieving data (Error: [Object Object])

DiedB, 2019-04-21:
Thanks for the report, will look into it!

DiedB, 2019-04-24:
Can you try v3.0.8 alpha?

RobKoenis, 2019-04-24:
In the appstore if i want to switch to alpha i get 'resource not found' message.

HansBouwman, 2019-04-24:
I've downloaded 3.0.8 for my Enphase system and it gives an error message stating:  Device not available. Error retrieving data (TypeError: ProductionData.reduce is not a function)

RobKoenis, 2019-04-25:
I have the same error now.
Before energie production was started this morning i did not have this error and the values where 0 as it should be.

DiedB, 2019-04-25:
Try v3.0.9 alpha, this one should finally be working ;-)

RobKoenis, 2019-04-25:
3.0.9, error was stil there. 
Removed the device in homey and added it again. Then it seems to work. 
Will see if it still works when the sun is down or a new day.

DDyerson, 2019-04-26:
how do i get the 3.0.9 version i have the same error(error retrieving data) 
this version isn't in de Homey app store yet!?!?

RobKoenis, 2019-04-26:
Yesterday in the evening it did not go to 0W but stayed at 4W.
Today when i looked in the homey app around 11:00 am it did still show the 4W and the KWh of yesterday. It showed that it was last updated 15 ours ago. Checking it some small time later it did still show thowe old values. But now at 13:30 when looking in the homey app it is showing the current values correct. 
Looking at homey Insights it seems the value was updated correct in the morning but somehow it was not updated in the homey app!
In Insight i can see it did stay at 4W until 12:00. Then it was 0W when starting a new day. So it does not go to zero when the sun goes down but keeps the last value.

DiedB, 2019-04-28:
Hmm, the app is pushing 0W as value when the Enphase API does not give any other information, which happens from midnight until production starts again. I could detect a production stop by parsing the timestamps and setting a treshold. Do you have any use case for this?

3.0.9 has been published which should at least get all Enphase systems working again. Finally...

RobKoenis, 2019-04-28:
The use case, at the end of the day when the sun is down.
Now it keeps showing the last watts reported but that could be of ours ago. This last value is normally very low (<10). But still not the actual value. Datetime parsing would be the solution. If the last reported time is older than ... then the value can be set to zero. Maybe 15 or 20 minutes!

HansBouwman, 2019-05-01:
I've monitored it for a few days and the 3.0.9 version works fine now for Enphase. At midnight the energy production value drops to 0 and the reported value correspond with the Enphase Enlighten App. Thanks for the effort.

DDyerson, 2019-05-01:
> how do i get the 3.0.9 version i have the same error(error retrieving data)
> this version isn't in de Homey app store yet!?!?

every thing works fine now!!
thnxs a milion

HansBouwman, 2019-05-01:
For sure the app is available in the Homey Store now. Just remove the device, the remove the old app, reinstall the 3.0.9 app and add the device. It should work then.

closed 2019-05-27 by DiedB