DiedB/Homey-SolarPanels #40: Solarman dual inverters - only updates one

LMSFan72, 2019-02-25:
Hi, I have two inverters on my system (one converting the solar to the mains and the other converting for battery storage). When I install SolarMAN and I log in with site ID I see both inverters and I select them both and they are then both visible in Homey (running Homey V2 and SolarPanels beta v2.4.0). However, only one inverter updates in Homey. If I then delete the one that does not update and reinstall it then that one updates but the other now stops! It appears that the app can only read one inverter.

LMSFan72, 2019-02-26:
I had an idea to work round this by creating a second site on my solarman account and having one inverter on each site. However, this still does not work - only the last inverter to be created updates.

DiedB, 2019-02-26:
This has been an issue across multiple inverter brands and I suspect that it has to do with the Homey software. It has never been reported before Homey v2 released.

LMSFan72, 2019-02-26:
Thanks Diederick. It's a shame because your app is perfect for what I wanted! I did have the following idea for a workaround:

Is it possible to create a "second" instance of the app and install that as well and see if one inverter can be linked to one, and the second to  the other. Unfortunately, I don't know how to do that but I envisage something like "SolarPanels2ndInverter" being the second app. In that case I would have "SolarPanels" and "SolarPanels2ndInverter" installed on Homey.... it's a bit clunky but it might be worth a try

DiedB, 2019-03-06:
Homey v2.0.5-rc.3 Experimental has a line in the changelog that might fix this issue:

> [Apps SDK] Fixed a bug where CronTasks would fire the wrong event

Could you please install that version and verify whether this fixes your issue? If it doesn't, please try to remove the inverters and the app entirely and recreate them again. Please report back!

LMSFan72, 2019-03-08:
Hi Diederick. How would I go about installing the Experimental version?

DiedB, 2019-03-08:
Looked through the settings but could not find anything that enables experimental updates. I think it's best to just wait it out, will probably take less than a week to get it to stable.

DiedB, 2019-03-20:
Version 2.0.5 should incorporate this fix. Can you install it and check if it solves your issue?

LMSFan72, 2019-03-20:
Hi, unfortunately that appears to have broken it completely...

Homey had automatically installed the update but neither device had updated their power readings since it had restarted. 

Therefore, I uninstalled both solarman devices and restarted again.

Now when I try to install a new device I get a timeout at 30000ms and cannot create any solarman devices..

DiedB, 2019-03-20:
Do you have their phone app as well? Does that work and is it stable?

LMSFan72, 2019-03-20:
Hi, yes, I am getting stable accurate readings through the solar man app and webpage.

LMSFan72, 2019-03-20:
Interestingly, I tried throughout today and had the same problem until just now. They have now both installed. I will see if they keep updating in the morning....

DiedB, 2019-03-20:
The APIs of these Chinese manufacturers seem iffy at best, please report back.

LMSFan72, 2019-03-21:
12 hours in and so far, so good!

DiedB, 2019-03-21:
I'll close this issue, as it was probably a problem on Athom's part.

closed 2019-03-21 by DiedB