DiedB/Homey-SolarPanels #41: Unknown_error when adding an enphase system

innacoon, 2019-03-01:
Beste Diederik,

Ik probeer mijn envoy-S meterd met homey te verbinden. Ik heb een developer account aangemaakt en deze toegang gegeven tot mijn enlighten. Hierna heb ik op homey een appartaat toegevoegd en de app geselecteerd, hier het nummer van de envoy-s (12182805xxxx), mijn API key en mijn enlighten ID ingevuld. Nadat ik op connect had gedrukt kreeg ik een pop-up met de te tekst “error: er is een onbekende fout opgetreden [unknown_error]”

Ik hoop dat je mij hierme kan helpen om toch verbinding te krijgen en jou app je kunnen gebruiken.

Ik gebruik overgens de Beta versie van de app, gezien enkel deze nog geinstalleerd kan worden.

RobKoenis, 2019-03-02:
Het nummer wat jij noemt lijkt op het serienummer van je envoy. Die heb je niet nodig. 

Systemid:
Login op 
https://enlighten.enphaseenergy.com/
Als je bent ingelogd en je systeem zie zal je URL iets zijn als
https://enlighten.enphaseenergy.com/pv/systems/1234567/overview
Het nummer in de URL is je systeem id.

Api key:
De api key die je bij developer.enphase.com heb aangemaakt. 
Bij de applicatie waar de api key staat heb je ook een authorization URL. Zorg ervoor dat je deze ook hebt aangeklikt om de applicatie die je hebt aangemaakt ook toestemming tot jouw systeem te krijgen. 

Userid:
Als je de app aan jouw systeem koppel  zoals bij de api key genoemd zal je ook je gebruikers id wel zien. 
Weet je deze niet (of niet zeker) log dan in op 
https://enlighten.enphaseenergy.com
Ga vervolgens naar instellingen. Onder api-instellingen zie je dan je gebruikers id die je gebruiken moet. 
Ook zie je een link 'Toepassingen weergeven die over Api-toegang beschikken. Klik daar op en kijk of je applicatie de je aangemaakt hebt op developer.enphase.com er tussen staat.

DiedB, 2019-03-20:
Did you succeed in adding your Enphase system, @innacoon?

innacoon, 2019-03-20:
I did succeed to add my Enphase system, in some time I will sort out how to implement the data into Homey.

Thanks for the reply.

Kind Regards, Met vriendeljke groet,
Eric Vrugt

Nobellaan 55
2641XV  Pijnacker
Tel:0648357688
eric_vrugt@icloud.com

closed 2019-03-20 by DiedB