DiedB/Homey-SolarPanels #45: Can not pair my omnik anymore

canedje, 2019-04-01:
I did remove my omnik solar system from Homey and wanted it to RE-pair it again. Because the energy was not updating
I’m not able to do pair it anymore.
It did work before.
I’m using 2.5 betha version  of the app and 2.1.0 Homey version

DiedB, 2019-04-02:
Are you still experiencing this issue or might it have been a temporary downtime?

DiedB, 2019-04-02:
Just checked, it is indeed timing out and their API seems to be down. Does your Omnik app still work?

canedje, 2019-04-03:
Yes the app still working
At the Homey forum more people are heading this problem

DiedB, 2019-04-03:
I guess I should be reading along there. Can you link users with the problems to this issue?

canedje, 2019-04-03:
https://community.athom.com/t/add-omnik-solar-system-results-in-timeout/8285

canedje, 2019-04-06:
Already some news?
Did you find the error?

nielsvn92, 2019-04-09:
I have the same issue as described above. I had the Omnik inverter connected for months, and around a month or so ago it suddenly stopped updating.
I just tried to reconnect my inverter, but I can't seem to add it.

Initially I got the error `Could not find that PairSession`. After I restarted the app, I got the time-out error.

Rlinsen, 2019-04-10:
hello, I haven't been able to connect for a couple of weeks. I tried both versions of the app, with no results. It seems as if the Connect button is not working.
I would like to hear if there is a solution for the problem.
Kind regards, Ronald

DiedB, 2019-04-11:
I will need someone to trace the app's communication. Maybe that provides a new endpoint we can use for Omnik inverters in this app. Until then, I am afraid that I cannot do much.

Rlinsen, 2019-04-11:
What can I do for you?

nielsvn92, 2019-04-11:
You could try and sniff the packets that are being sent by the Omnik app on your phone.
I'll try and do it too, but that'll be tomorrow I guess.

DiedB, 2019-04-11:
That's what I mean. For me it's quite hard to do, as I am using an iPhone (does not allow me to record traces) and I do not have an Omnik account handy.

nielsvn92, 2019-04-11:
Ok what I did was running Fiddler on my laptop, and followed these instructions: https://docs.telerik.com/fiddler/Configure-Fiddler/Tasks/ConfigureForAndroid.

Don't forget to configure Fiddler to allow remote connections, and decrypt HTTPS traffic (it will install it's own certificate on your computer..).

After that, I opened the app and logged in. It seems t hey changed their API..
The following calls were done:
**Request**
```POST https://api.omnikportal.com/v1/user/account_validate 
uid: <user ID>
appid: 10038
appkey: <shown, but I'll skip it for security reasons>
Content-Type: application/x-www-form-urlencoded

user_email=<email>&user_password=<password>&user_type=-1``` 

**Response**
```{"error_msg":"","data":{"c_user_id":<user ID>,"user_type":"1","plant_count":1},"error_code":0}```

**Request**
```GET https://api.omnikportal.com/v1/plant/list?page=1&perpage=20&locale=en-US HTTP/1.1
uid: <user ID>
appid: 10038
appkey: <shown, but I'll skip it for security reasons>
Content-Type: application/x-www-form-urlencoded
```

**Response**
```{"error_msg":"","data":{"plants":[{"total_energy":"1210.10","plant_id":<plant ID>,"country":"Netherlands","city":"<city>","timezone":"1","image_url":"http://www.omnikportal.com/images/UserImages/Default.jpg","latitude":"<latitude>","line_status":"1","peak_power":3.475,"installer":"<installer>","user_id":<user ID>,"alarm_status":"","name":"<name>","eday":"4.01","currency":"EUR","create_date":"2018-09-07","longitude":"<longitude>","operator":"","locale":"en-US","income":"<double>","current_power":2.72,"today_energy":4.01,"status":1}],"count":1},"error_code":0}```

**Request**
```GET https://api.omnikportal.com/v1/plant/data?plant_id=<plant ID>HTTP/1.1
uid: <user ID>
appid: 10038
appkey: <shown, but I'll skip it for security reasons>
Content-Type: application/x-www-form-urlencoded
```

**Response**
```{"error_msg":"","data":{"total_energy":"1210.10","last_update_time":"2019-04-11T09:16:46Z","peak_power_actual":3.019,"timezone":"+01:00","currency":"EUR","today_energy":4.01,"monthly_energy":145.71,"yearly_energy":644.2,"current_power":2.68,"income":"242.02","carbon_offset":1.21,"efficiency":1.56},"error_code":0}```

nielsvn92, 2019-04-11:
Oh I also see an endpoint to list the devices on a plant:

**Request**
```
GET https://api.omnikportal.com/v1/device/list?plant_id=<plant ID>&page=1 HTTP/1.1
uid: <user ID>
appid: 10038
appkey: <key>
Content-Type: application/x-www-form-urlencoded
```

**Response**
```
{"error_msg":"","data":{"devices":[{"total_energy":"1212.80","last_update_time":"2019-04-11T10:21:53Z","device_sn":"<inverter serial number>","alarm_status":"0","sensor":"0102","today_energy":"8.07","line_status":"1","power":"2830","datalogger_sn":"<datalogger serial number>","device_id":<integer>,"status":1,"manufacturer":"Omnik","model":"","type":1}],"count":1},"error_code":0}
```

DiedB, 2019-04-11:
Just tried it using my own Omnik account (does not have associated plants). At the user validation stage, I get an error, presumably because the combination of appid and appkey are not valid. This is probably a mechanism set in place by Omnik to only allow their authorized apps to connect.

The question is whether the appid and appkey are static, either at user level or at application level. Then again, no one knows when they will change.

Can you reinstall the app and check whether the appid and appkey are the same? If so, please post them so I can have a look.

nielsvn92, 2019-04-11:
For now it uses the same appid and appkey. I'm not sure if I should share the appkey here in Github.
Do you have any other medium where I can send it to?

We indeed don't know when the key will change, will have to find that out 😁 .

DiedB, 2019-04-11:
If we were to go this route, the app key would become public anyway (since it would be part of the source code), so you might as well post it here. I will try and implement it in the newest app version (3.0 beta), but I'm a little worried about it breaking again.

nielsvn92, 2019-04-11:
Ah true, you're right. We'll see what happens, great to hear that you want to implement it! :-)
```appid: 10038
appkey: Ox7yu3Eivicheinguth9ef9kohngo9oo```

DiedB, 2019-04-11:
Nice, it seems to be working with my own account as well. I'll keep you posted!

nielsvn92, 2019-04-11:
Awesome! If I can do anything to help, please say so :-).

Rlinsen, 2019-04-11:
hello, i don't quite understand what is being told here. am not familiar with program code, but an ordinary user. Can I use this information to do something to link the app? or do I have to wait for version 3.0? I like to hear what I can do best.
Greetings, Ronald

nielsvn92, 2019-04-12:
> hello, i don't quite understand what is being told here. am not familiar with program code, but an ordinary user. Can I use this information to do something to link the app? or do I have to wait for version 3.0? I like to hear what I can do best.
> Greetings, Ronald

You'll have to wait until version 3.0 beta.

DiedB, 2019-04-12:
3.0.1 beta, to be precise ;-)

canedje, 2019-04-12:
I just installed v3.0.0
This version is missing Omnik as brand

DiedB, 2019-04-12:
As I said, Omnik will be added in 3.0.1 beta.

Rlinsen, 2019-04-12:
Hello thank you
I am waiting for the next version for my Omnik to be able to use the app again with pleasure.
Greetings, Ronald

DiedB, 2019-04-13:
Support has been added in v3.0.1, should be in the app store (as beta) somewhere in the next few days.

closed 2019-04-13 by DiedB

Rlinsen, 2019-04-16:
Diederik, Version 3.0.2 installed. it seemed to work fine again. But now I get the message: error retrieving data (fetcherror: invalid json response body at https://api.omnikportal.com/v1/user/account_validate reason: unexpected token <in json at position 0)
Can you do something with this?
Ronald

nielsvn92, 2019-04-16:
Confirmed, I have the same issue. It seems to be incidental, but the device doesn't seem to recover if this happens. It's strange, because the call only needs to be done once when pairing. Afterwards its not needed anymore because you have the user ID.
Could this be the culprit?