DiedB/Homey-SolarPanels #48: After update SE fails (403) and API key gone

wfdekker, 2019-04-13:
After the auto upgrade the SolarEdge gives error and when I looked in settings the API key was gone. After adding the key again it still doesn’t work, looks like all settings have been purged?

DiedB, 2019-04-13:
Can you try removing the SolarEdge device, then removing the app, then reinstalling the app and adding back the SolarEdge inverter? Version 3 uses different capabilities and auto-upgrade will not be working.

kivimart, 2019-04-13:
Data is coming in after delete and reinstall app. added the solar panel againg.

ncickc, 2019-04-13:
But can't create flows where the solar system is the trigger (app not visible in the 'if' column)

DiedB, 2019-04-13:
Has been fixed in v3.0.2, which will yet again require you to remove and re-add your inverter to Homey. Sorry for that, the beta version got accidentally pushed to the stable branch.

DiedB, 2019-04-13:
That update has been submitted to Athom but might take a few days to be published.

closed 2019-04-13 by DiedB

wfdekker, 2019-04-13:
It worked already again after afding the API key again, but if I delete everything I have to fix/recreate all the flows I use the inverter values in right? Prefer not to do that while it still seems to work. 
I will wait till the 3.0.2 is out and fix it by removing and reinstalling if required.