DiedB/Homey-SolarPanels #53: Solaredge doesn't work anymore in 3.0.0

Bakkerwj, 2019-04-14:
Solaredge was working perfectly but since update 3.0.0 it doesn't do anything anymore. sometime's it works shortly but give's info from 2 day's ago.

Robar56, 2019-04-14:
Same issue, 
When trying to re- install PV-Output, after the update, after given the api-key, an error is given and the homey-app stopped.
After restart the homey-app, the PV-output card is seen, but, not when I try to recover my flows, the PV-Output is not in that list.

corvdniet, 2019-04-14:
Same issue. I reinstalled the app, added a device and then Homey app stops. The newly added device shows up in the device list. But you can't add a flow card ... it simply does not show up in the possible flowcards list. Mine is a SolarEdge device.

Bakkerwj, 2019-04-14:
For me also no flow cards ...

DiedB, 2019-04-15:
Please wait until v3.0.2 (is under approval by Athom, should be released today or tomorrow). When the update has been installed, please remove your SolarEdge devices, remove the app, then reinstall the app and add your devices again. That should fix all problems (including the flow cards). Sorry for the inconvenience.

DiedB, 2019-04-16:
v3.0.2 has just been published. Can you try reinstalling the app and your inverters, and check whether everything is working fine again?

Bakkerwj, 2019-04-16:
[image]

Not working......

DiedB, 2019-04-16:
How long ago have you added the system? It can take up to 20 minutes for it to first get its data.

Bakkerwj, 2019-04-16:
Indeed, i was too fast… Now it works, but took about 40 min…
In what interval is it updated? every??

Met vriendelijke groet,

Willem Jan Bakker
Boerestreek 1
8757JR Gaast
Mob. +31653217707

DiedB, 2019-04-16:
Every 10 minutes, given that SolarEdge has new data at that point. It's a limit on the SolarEdge side, they do not allow many API calls.

Be advised that GitHub makes this conversation public, you might want to remove your signature.

Robar56, 2019-04-16:
Hoi Diederik,

Dank voor je werk. PVOutput werkt weer als vanouds.

Groet

Rob

Bakkerwj, 2019-04-16:
Thank you! works great again... nog my sunscreens go down again...😎

interested in making an app for Nibe?

https://api.nibeuplink.com

DiedB, 2019-04-16:
@Bakkerwj Sounds interesting, Homey does not support any heat pumps yet as far as I can see. Send me an email at diederik@outlook.com and I will look into it.

Writing apps for devices with well-documented APIs isn't a lot of work.

closed 2019-04-16 by DiedB

corvdniet, 2019-04-16:
Dank je wel Diederik