DiedB/Homey-SolarPanels #56: Omnik time out v3.02

canedje, 2019-04-16:
The app is timing out again at Omnik
It did work for an hour before

DiedB, 2019-04-16:
I am aware of this, not sure what the cause is yet. Might be a rate limit that is hit, since it seems to happen every day around the same time. Will be creating a fix.

canedje, 2019-04-16:
After an hour again a red triangle.
It is saying something like: fetcherror unexpected token in JSON at position 0

Rlinsen, 2019-04-16:
Diederik, Version 3.0.2 installed. it seemed to work fine again (omnik). But after about an hour I get the message: error retrieving data (fetcherror: invalid json response body at https://api.omnikportal.com/v1/user/account_validate reason: unexpected token <in json at position 0)
Can you do something with this?
Ronald

DiedB, 2019-04-16:
I have created an implementation with some more robust error handling. I'll run it for a few days and see if it goes wrong again.

DiedB, 2019-04-19:
Please try v3.0.6 and see if it fixes the issues.

closed 2019-04-19 by DiedB

canedje, 2019-04-19:
Hallo Diederik ,

It worked for Some time.
But at the end the red triangle.
The device is giving error: Error appid folowed by a Chinese sign

But after a couple off minutes the device is recovered again and is up in the air
 

Mvg,
Edward Hogeveen

Verstuurd vanaf mijn iPad

canedje, 2019-04-19:
So there is sure a big improvement!!!

Mvg,
Edward Hogeveen

Verstuurd vanaf mijn iPad