DiedB/Homey-SolarPanels #57: trannergy disappeared

bobmijwaard, 2019-04-16:
I have installed the latest version of solar panels but trannergy seems to be disappeared. Will it be back in a later version?

DiedB, 2019-04-16:
Will be back soon!

bobmijwaard, 2019-05-30:
Any news on adding Trannergy again?

DiedB, 2019-05-30:
Soon was a little optimistic, but I will find/make time for it sooner or later.

MaxvandeLaar, 2019-12-01:
@DiedB 
Why did you removed it and is there a way I can help with this?

DiedB, 2019-12-01:
It was removed in the migration to Homey v3.0 probably. Can you send me your Trannergy account details at diederik@outlook.com? I'd like to re-implement it but I need credentials for testing.

stephanhaas, 2019-12-02:
@DiedB If you're in need of a Trannergy account I could help you. Would really like Trannergy added to your app.

DiedB, 2019-12-02:
Sure!

stephanhaas, 2019-12-12:
@DiedB just to be sure, did you revive my email?

DiedB, 2019-12-12:
I did!

stephanhaas, 2019-12-12:
OK, thanks.

MaxvandeLaar, 2020-02-23:
@DiedB is there any progress on this?

DiedB, 2020-03-05:
Not really, I don't have an account and API documentation. If I remember correctly, Trannergy support was dropped when they changed their API two years ago.

MaxvandeLaar, 2020-03-06:
@DiedB, sooo I really want Trannergy added. I decompiled an android apk and figured out their API. Do you accept pull requests so I can try to add it myself?

stephanhaas, 2020-03-06:
interested!

DiedB, 2020-03-06:
@MaxvandeLaar Cool, and I sure do! Please have a look at the other inverter implementations and try to not stray away too much, keeps the app maintainable ;-) Good luck and let me know here if you need help.

MaxvandeLaar, 2020-03-06:
> @MaxvandeLaar Cool, and I sure do! Please have a look at the other inverter implementations and try to not stray away too much, keeps the app maintainable ;-) Good luck and let me know here if you need help.

Great. I noticed SolarMan uses the same api (except base url) so I just copied it all and changed the url and some labels and icons. I currently installed it on my Homey and it connected without issues. I will probably make a pull request somewhere this weekend.

DiedB, 2020-03-10:
For everyone who wants to test along, please install the beta version using https://homey.app/a/it.diederik.solar/test/ and report back with your findings!

stephanhaas, 2020-03-13:
Installed the beta version and it seems to work. When looking at insights the total generated power of yesterday looks to be reset at 07:20 today. Perhaps in the coming days this will sort itself out. Don't know what the intended behaviour should be.

DiedB, 2020-03-13:
That is intended in some way, it just takes over the daily generation value from the Trannergy API. This resets early in the morning, apparently.

stephanhaas, 2020-03-13:
OK, thanks. I'll adjust my flow to a calculation early morning.

DiedB, 2020-03-18:
Has been added to the live version, thanks @MaxvandeLaar!

closed 2020-03-18 by DiedB