DiedB/Homey-SolarPanels #61: S

giggo, 2019-04-29:


giggo, 2019-04-29:
Hi
do you plan to support the "Sonnen" Battery?

https://www.loxforum.com/filedata/fetch?id=58670

http://192.168.1.10/api/v1/status

Example output:
---
{"BatteryCharging":false,"BatteryDischarging":false,"Consumption_W":480,"Fac":50,"FlowConsumptionBattery":false,"FlowConsumptionGrid":true,"FlowConsumptionProduction":false,"FlowGridBattery":false,"FlowProductionBattery":false,"FlowProductionGrid":false,"GridFeedIn_W":-485,"IsSystemInstalled":1,"Pac_total_W":-4,"Production_W":0,"RSOC":5,"Timestamp":"2019-04-30 00:45:37","USOC":0,"Uac":229,"Ubat":48}
---

Ciao,
gagga

DiedB, 2019-04-30:
The Lox Forum link is not working.

Connecting batteries to Homey sounds great, but I do not think it fits in this app (arguably, it is already too big). Another app would have to be created, dedicated to batteries or even dedicated to just the Sonnen brand.

giggo, 2019-05-24:
The Sonnen battery API provides not just battery data but inverter data too. The battery controller connects to any of their supported inverters (pretty much all at the market) so supporting the battery API gives support for many other inverter systems too.

This is what the parameters actually mean:

Consumption_W House comsumption in watts
Production_W PV Production in watts
Pac_total_W Inverter AC Power greater than ZERO is discharging, Inverter AC Power less than ZERO is charging
RSOC Relative state of charge
USOC User state of charge
Fac AC frequency in hertz.
Uac AC voltage in volts
Ubat Battery volatge in volts
Timestamp System time
IsSystemInstalled System is installed or not

DiedB, 2019-05-27:
I think it would be a good idea to implement this into a separate app, which would be able to support battery systems (Tesla Powerwall, Maxem, Sonnen). I am willing to spend some time on this, but I would need users from those other systems as well.

LMSFan72, 2019-05-30:
I have a SOFAR ME3000SP which talks through the SolarMAN API I believe. Happy to help test that. I would be looking for battery charge/discharge power and remaining capacity