DiedB/Homey-SolarPanels #62: Steca support

DiedB, 2019-04-30:
https://community.athom.com/t/solar-panels/12068

ArnfinnGilberg, 2019-05-02:
I would really like to get my StecaGrid inverter included in this app.
This is a link to my dashboard page, don't know if this helps:
https://public.meteocontrol.de/dashboard/system/LQCNE/GH4Bbe49m1?lang=en
https://public.meteocontrol.de

Best regards Arnfinn

DiedB, 2019-05-02:
We can read from that API. Do all Steca users have such a dashboard available? Does Steca offer an app? What did you do to get that dashboard for your system? Is it always public?

https://ws.meteocontrol.de/api/int/dashboard/v1/sites/LQCNE/data/energygeneration?apiKey=GH4Bbe49m1&lang=en&type=day&date=2019-05-02&_=1556828289306

ArnfinnGilberg, 2019-05-02:
Great!

To my knowledge, yes, all Steca get a 2 year free access to their own site at the metocontrol.de page.
After the grid inverter was set operational I made a profile at the metocontrol.de page with the serielnumber from the inverter.

To my knowledge it is public, but I can access my site and fill in site name and a few details.
No app available.

Thanks
Arnfinn

JRTimmerman, 2019-10-25:
Besides the above mentioned site, a lifetime connection is available through 
https://steca.powerdoo.com/

here the logging is automatically send to.

evhetlan, 2023-11-15:
New link: sunCloud.steca.com