DiedB/Homey-SolarPanels #65: Growatt could not login

RosRoos, 2019-05-23:
Hello,

I'm happy that the app is supporting my growatt system.
But when i add an new device, select growatt an fill in my credentials it gives me the error:
Could not login to Growatt server.

Did i forget something?
Thanks,
Robin

pro-sumer, 2019-05-23:
Hello Robin,

I wrote the driver for Growatt plugins. Hope I can help you.

Do your credentials work on https://server.growatt.com?

RosRoos, 2019-05-23:
thanks!
yes im currently logged in https://server.growatt.com/panel.do
so my credentials are working

Roylegr, 2019-05-23:
Same here, i've the same error.
But by the server i need to verify with 3 letters or numbers

pro-sumer, 2019-05-23:
Nice to meet two fellow Growatt users!

I'm sorry that the driver does not work for you. Let's try to fix that together.

Can you please install the [HomeyScript App](https://apps.athom.com/app/com.athom.homeyscript)?

Then open the [online editor](https://homeyscript.homey.app/) and create a new script with this content:

```
let result = await fetch('https://server.growatt.com/newLoginAPI.do', 
  { method: 'POST', 
    headers: { 'Content-Type': 'application/x-www-form-urlencoded' },
    body: 'userName=USERNAME&password=HASH'
  }
)
let body = await result.json()
return body.back.success
```

In this script you have to replace two "placeholders":
* `USERNAME`: your Growatt username
* `HASH`: a hash of your Growatt password

The hash is a slightly modified [MD5](https://nl.wikipedia.org/wiki/MD5) hash of your password:
1. Split the regular MD5 hash up in pairs of 2 characters
2. For each pair replace the first character by a `c` if it is a `0`
3. Join the pairs again

If necessary use an online [MD5 Hash Generator](http://onlinemd5.com/) (if you are comfortable typing your password on such a site; I'd rather do it on a local system myself).

Example:
* password is `test`
* MD5 hash is `0cbc6611f5540bd0809a388dc95a615b`
* Pairs are `0c bc 66 11 f5 54 0b d0 80 9a 38 8d c9 5a 61 5b`
* Result after replacing is `cc bc 66 11 f5 54 cb d0 80 9a 38 8d c9 5a 61 5b`
* String to use for `HASH` is `ccbc6611f554cbd0809a388dc95a615b`

If the hash calculation is too complicated you can also temporarily change your Growatt password to `test` and use the hash `ccbc6611f554cbd0809a388dc95a615b`.

Please post the result of running this script, which should either be `true` (success!) or `false` (login fails).

pro-sumer, 2019-05-23:
@DiedB Feel free to assign this issue to me.

RosRoos, 2019-05-24:
thanks for your clear explanation.
i got:

----------------
Script returned:
 false

i use as username my email address..

pro-sumer, 2019-05-24:
That's at least consistent with the failed attempt from the SolarPanels App.

Can you try the script again with this return statement instead? (remove `.back.success`)

```
return body
```

That should include an error code (though it will probably be 502 and not tell us much).

Is the extra account you created a dummy one which credentials you can share here?

RosRoos, 2019-05-24:
indeed i got:

----------------
Script returned:
 { back: { success: false, msg: '502' } }

you can use this extra account: robinros
password: test

can you give me a reply when you are done so i can change the pw :)

pro-sumer, 2019-05-24:
Somehow I can't use HomeyScript right now; it complains that my Homey is offline...

I briefly tried this:
* log in on the website: works fine
* log in using a REST client: fails with error code 502 (might mean the hash is wrong)

Seems like the site I suggested for creating an MD5 is giving incorrect results...
I created the hash locally on Python and indeed got a completely different result!

Can you please try again with this hash `c98f6bcd4621d373cade4e832627b4f6` for password `test`?

I will update my post with instructions.

RosRoos, 2019-05-24:
great! i got this result:

----------------
Script returned:
 { back: 
   { data: [ [Object] ],
     service: '1',
     quality: '0',
     isOpenSmartFamily: 0,
     totalData: {},
     success: true,
     user: 
      { uid: '',
        userLanguage: 'ho',
        inverterGroup: [],
        timeZone: 1,
        lng: '',
        dataAcqList: [],
        type: 0,
        password: 'c98f6bcd4621d373cade4e832627b4f6',
        isValiPhone: 0,
        kind: 0,
        mailNotice: false,
        id: 294406,
        lastLoginIp: '100.116.18.44',
        phoneNum: '',
        approved: false,
        area: 'Europe',
        smsNotice: false,
        isAgent: 0,
        token: '',
        nickName: '',
        parentUserId: 186720,
        customerCode: '',
        counrty: 'Netherlands',
        isPhoneNumReg: 0,
        createDate: '2019-05-24 14:40:23',
        rightlevel: 2,
        appType: '',
        serverUrl: '',
        lat: '',
        lastLoginTime: '2019-05-24 17:38:01',
        enabled: true,
        agentCode: '',
        inverterList: [],
        isValiEmail: 0,
        accountName: 'robinros',
        email: 'robin@robinros.nl',
        company: '',
        activeName: '',
        codeIndex: 1,
        appAlias: 'robinros',
        isBigCustomer: 0,
        noticeType: '' },
     msg: '',
     app_code: '8' } }

pro-sumer, 2019-05-24:
Yes; this login succeeded.

It might be wise to remove some personal info, like your email address, from this post.

Can you try two more things now and post the results?

1. Log in using the Homey SolarPanels App with password `test`
2. Change your password back to the original and try that (with the new MD5) in HomeyScript

RosRoos, 2019-05-24:
i can now login with test as password.
i changed my password, removed the deviced.
logged in again and its working with my new password!?

pro-sumer, 2019-05-24:
Weird. So you can use the SolarPanels App now, with your "real" password?

Maybe @Roylegr can test this as well?

RosRoos, 2019-05-24:
yes its working now for me :)!

Roylegr, 2019-05-24:
Hey,
I tested it as described.
"True"
But can't log in.
"Could not find that PairSession"
Maybe later when the sun shines again?

pro-sumer, 2019-05-24:
I don't recognize the "Could not find that PairSession" error, but apparently this also happened with the driver for Enphase inverters:

https://community.athom.com/t/enphase-zonnepanelen-toevoegen/9609

The driver I wrote only communicates with the Growatt website; not with the inverter itself. Hence, whether the sun shines (for my inverter: whether it is powered) should not matter.

Did you test with your own password or with `test` in HomeyScript?

Roylegr, 2019-05-25:
No i had tested only with my own password.
But will test later with 'test'

Roylegr, 2019-05-25:
Okay,
I've been testing something. It works with the password "test". Although I did not understand that I received a different HASH but that was because the HASH is "Test" and not "test" but it works.
Password "test" and HASH "Test".
The script always gives "false", but it works.
With another password it doesn't work for me, capital letters or not HASH with capital letters or not. Don't get it working. Script gives "true" but does not work.
Get the password still working and it's a great app !!!!! (now also)

pro-sumer, 2019-05-25:
@Roylegr I’m not sure I understand your last post.

What’s your current status?

- Is the SolarPanels App with the Growatt plugin now working for you as well?
- With a different password than “test”?

Roylegr, 2019-05-25:
no it only works with "test". Don't get it working with another password even without numbers or symbols.
I replace the HASH with my password and the 0 in C in the HASH. On the Growatt server and in the app but that doesn't work.

pro-sumer, 2019-05-25:
@Roylegr `HASH` should be the `md5` hash of your password (with some `0`’s replaced by `c`), not the password itself.

Did you calculate the `md5` hash of your password?

Note that this hash is for testing in HomeyScript only.

You should be able to use your regular password for the Growatt server in the SolarPanels App as well (the driver calculates the hash for your password).

I still don’t understand why this (initially) does not work for both of you.

<sub>Sent with <a href="http://githawk.com">GitHawk</a></sub>

Fire69, 2019-05-25:
I'm having the same problem in the app.
Haven't tried with HomeyScript because I can't change the password (I'm testing with my dad's account until we get our own panels installed :P )

Roylegr, 2019-05-25:
okay strange, I have replaced the password again and first logged out at the growatt server and logged in again.
then replace the password in the app. Now it works !! but a 2-digit password does not work

pro-sumer, 2019-05-25:
@Fire69 You don’t have to change the password to test with HomeyScript. You should be able to test with your current password.

However, it somehow looks like changing the password triggers something that solves the initial login problem...

<sub>Sent with <a href="http://githawk.com">GitHawk</a></sub>

martijngeerts, 2019-05-28:
I have the same problem, it only works with the md5 hash of "test" if i change my password the scrips returned as true. in the app i get: Could not find that pairsession.

pro-sumer, 2019-05-28:
@martijngeerts I have never encountered the `PairSession` error myself.

When did you start getting this error? And do you always get it, or under certain conditions? (Which?)

martijngeerts, 2019-05-28:
Mmm i tried it againg (reset) and now it's "could not login to Growatt server" sometimes i get the Pairsession.
I only did succeed to login with the "test" password.

pro-sumer, 2019-05-28:
There should be nothing special about password `test`.

Did you try several others?

Can you please post one that did not work? And the hash that you calculated for that password?

martijngeerts, 2019-05-28:
i tried many others. example password: hapsnurk with the calculated hash: 64d4832469b0d59d81e73c17e78232b4 

script returned as true but does not work in the app

martijngeerts, 2019-05-28:
i tried another one and this time i replaced all the 0 to c and it's working now :) very strange

pro-sumer, 2019-05-28:
@martijngeerts Thank you very much for posting the password `hapsnurk`, as I noticed something interesting:
* manual calculation of hash leads to `64d4832469b0d59d81e73c17e78232b4`
* my code leads to `64d4832469bcd59d81e73c17e78232b4`

So a `0` got replaced by a `c`, but since this is the second digit in a pair, that should not have happened.

Looks like I was too optimistic when I thought I could implement this hash calculation as a one-liner using a regular expression...

You helped me find the/a root cause! I should be able to fix this (hopefully later this week).

DiedB, 2019-05-28:
Let me know when you have a pull request ready, and I will release a new version.

pro-sumer, 2019-05-29:
@DiedB I created [Pull Request 66](https://github.com/DiedB/Homey-SolarPanels/pull/66) to fix this.

PS: I got punished big time for (solving) this bug... After testing my fix on my Homey and re-installing the released App, Homey decided to delete *all* my SolarPanels data because they were not compatible?! 😢

(Is this because of the previous alpha group warning?)

DiedB, 2019-05-29:
Update has been pushed on the alpha channel.

commit 6af98195 (2019-05-29): Fixes #65 by replacing only the first `0` by a `c` in every pair of 2 digits in the md5 hash of the user's password
commit b6c48aa6 (2019-05-29): Merge pull request #66 from pro-sumer/master

Fixes #65 by replacing only the first `0` by a `c` in every digits pair
closed 2019-05-29 by DiedB