DiedB/Homey-SolarPanels #68: AP Systems support

upviral, 2019-06-24:
Hey Diederik :) 

Are you planning/considering to add support for AP Systems (see apsystemsema.com) support as well? Happy to give you temporary access while you're testing the integration, if needed.

Kind regards,

Wilco de Kreij

P.S. Beide Hollanders, maar dacht.. laat ik het maar in het Engels doen ;)

DiedB, 2019-06-26:
English is a good choice, as others can read along as well.

I think it would be possible, but they do not have official support for external applications. They do have their own apps. We can reverse engineer those and connect Homey to the same endpoints, but this solution has not always proven stable in the past. I cannot find any other integrations online, but I might miss something.

I'll keep this issue open and let others chime in. Once there is enough interest, I will see if I can find some time to give it a try.

Michielfs, 2020-04-17:
Hi Guys,

Just got a APsystems setup and noticed that i can not connect it to homey. Is there a chance that it will in the future? Hope so👍🏼

Greetings Michiel FS

DiedB, 2020-05-07:
https://medium.com/@rukmalf/extracting-data-from-apsystems-inverters-8c2b8e8942b6

Does this work for you?

Michielfs, 2020-05-12:
Hi,

Thanks! Looks great. I will try it soon and give a update.

gregoiredore, 2020-11-30:
That would be awesome ! thanks in advance for the amazing work

ElineSchaft, 2022-03-27:
Hi! I have the same setup. Is there a solution to connect it to Homey?

Thanks!!

codex666-cenotaph, 2022-06-01:
+1 for the aps integration. there is a thread on tweakers that discusses a current implementation on home-assistant. I hope it will be possible to reuse this info for a homey version.
https://gathering.tweakers.net/forum/list_messages/2032302/1

2tracer, 2022-10-05:
+1 on integration on Homey. 🤞🏻🤞🏻

robgBY, 2023-07-24:
Hi @DiedB,

I noticed this issue is very silent, but I wonder if supporting APS for Homey (Pro early 2023) is still planned?

monhro, 2023-09-22:
Hi folks, a +1 for APS integration ! :)

monhro, 2023-10-16:
Last update, it seems like Home Assistant supports AP Systems integration, could be worthy to take a look, could make this easier ?
https://github.com/ksheumaker/homeassistant-apsystems_ecur

cypress0815, 2023-10-31:
Another thumbs up for APSystems ECU-B integration

babjerke59, 2023-11-08:
+1 for APS integration !

Ruudje007, 2024-01-26:
+1 for as well

Huisweev, 2024-01-27:
I hope you will AP systems integrate for Homey

festeny, 2024-08-29:
+1 for releasing an APS Homey App that supports my APsystems ECU-R.

I also found this workaround via scripting, but it doesn't seem to for me: https://community.homey.app/t/solar-system-from-apsystems/74759/4?page=2

Would you be able to help out? Much appreciated!

pwvandeursen, 2024-08-29:
This one unfortunetely does not work for me (the script, also clear from other feedback in the same link). that being said, I have got it to work through HA and linking that then back to Homey.

2tracer, 2024-08-30:
+1. this must be done soon. 
APsystems loose huge marked shares to asia because they are so slow..

theohendriks85, 2025-01-09:
I've been busy with this for a while myself. I am now able to log in locally via the IP address on an ECU-R 2162 ** (this did not work for my ECU R 2160**). Then I read the ECU from a homey script. I can also turn them on and off with a webhook. If anyone is interested, let us know.

nielsbrakel, 2025-04-15:
@DiedB Can you shine some light on the status?

Dickje, 2025-04-18:
Hi Diederik, I'm in the process of making an app for APsystems, making use of their API. I have requested an API key and secret, and AP systems apparently has no problem with supplying that.
I've got a piece of code to retrieve data in JSON format. 
I don't want to reinvent the wheel, and since you already made an app; can it incorporate AP Systems and handle JSON responses from the API?

Best regards, Dick

Ruudje007, 2025-04-18:
You can retrieve your own secrets by visiting the following URL:
https://apsystemsema.com/apsystems/web/setting/personalSetting/openAPIService

Here you can also download the [manual](https://file.apsystemsema.com:8083/apsystems/resource/openapi/Apsystems_OpenAPI_User_Manual_End_User_EN.pdf )

Maybe this helps?

Cheers

Barblo1986, 2025-04-20:
+1 for the Homey integration with APSystem! Only my concern is that Diederik might no longer be actively maintaining the app 😞. Last update was in December 2023, and in the meantime, separate apps are already being developed to replace parts of this one.

robgBY, 2025-05-03:
> Hi Diederik, I'm in the process of making an app for APsystems, making use of their API. I have requested an API key and secret, and AP systems apparently has no problem with supplying that. I've [...]

Hi @Dickje I was wondering if you where able to get real data out of the APSystems API. A while ago I got API credentials myself. Through a piece of Python code I have been able to connect to the API successfully, but unfortunately I get stuck on a return code 4000: "Request parameter exception". I've seen others get stuck the same way.

Would be very interested to know if you got any further that that, and have been able to pull some real data from the API?

The Python code I used:

```
import uuid
import time
import hmac	
import hashlib
import base64
import requests

#Variabelen
base_url = "https://api.apsystemsema.com:9282"
nonce = str(uuid.uuid4()).replace('-','')
ts = int((time.time()*1000))
timestamp = str(ts)
sid = "xxxxx" use you SID

api_key = "xxxxx"  #use your key
api_secret = "xxxx" #use your secret

request_path = f"user/api/v2/systems/details/{sid}"
string_to_sign = f"{timestamp}/{nonce}/{api_key}/{request_path}/GET/HmacSHA256"
string_bytes = string_to_sign.encode('utf-8')
api_bytes = api_secret.encode('utf-8')
encrypted = hmac.new(api_bytes, msg = string_bytes, digestmod = hashlib.sha256).digest()
signature = str(base64.b64encode(encrypted)).replace('b','')

header = {
'X-CA-AppId': api_key,
'X-CA-Timestamp': timestamp,
'X-CA-Nonce': nonce,
'X-CA-Signature-Method': 'HmacSHA256',
'X-CA-Signature': signature
}

url = f"{base_url}/{request_path}"

response = requests.get(url = url, headers = header)
response.raise_for_status() # Controleer op HTTP-fouten
print(response)

if response.status_code == 200:
	print("Succes:", response.json())
	print("Headers:", response.headers)
	print(response.url)
	solar_data = response.json()
else:
	print("Fout:", response.status_code, response.text)
```

Dickje, 2025-05-03:
Hi RobgBY,

I did not get any good results with a Python script, but a JavaScript yielded results. My guess is that the signature is not calculated correctly in Python. By the way; in the ‘string_to_sign’, don’t put in the full request path, but only the part after the last /  in the request path, like here it would be ‘details’.
 Regards, Dick