DiedB/Homey-SolarPanels #69: Sar edge pair not working

Raycokes2000, 2019-07-01:
Hello, my api key and side Id is not working with my solar edge. It says can not login.
But that's all got from my engineer.

Gr ray

DiedB, 2019-07-01:
I remember this going wrong when the installer did not click 'save' after retrieving the API key.

Raycokes2000, 2019-07-01:
.??? Installer did not click save?

DiedB, 2019-07-01:
On the page where an installer can find the API key, you have to click 'save' before the API key actually starts working.

Raycokes2000, 2019-07-01:
There is no save button, only a connect button who generates the error mentioned earlier.

Raycokes2000, 2019-07-01:
Or I receive the error; could not find that PairSession.

DiedB, 2019-07-03:
You received the API key from the installer/engineer of the solar panels, right? I am suggesting that that is where it went wrong. When the engineer retrieves the API key, it is important to click save first. Do other apps work with your SolarEdge system.

Raycokes2000, 2019-07-03:
Aha, that's something else. Installer is software. Have no other apps then original solar edge and now the homey app. I guess the support of seller is not that stupid but I will ask to be sure.

Raycokes2000, 2019-07-04:
Seller checkt the key, all good he says.

DiedB, 2019-07-04:
Something must be wrong with the data you are entering, or more people would be complaining about SolarEdge not working here. The app has lots of SolarEdge users.

Raycokes2000, 2019-07-04:
Yep, I now also have solaredge support looking at it.

DiedB, 2019-07-13:
Any update on this issue?

Raycokes2000, 2019-07-13:
Resend it again, unfortunately no respons yet 😢

DiedB, 2019-07-27:
Still no response? It must be a problem on your side as no one else seems to have issues with SolarEdge.

closed 2020-03-07 by DiedB