DiedB/Homey-SolarPanels #70: Kostal Piko converter

Larnak23, 2019-07-20:
I would like to know if your are able to add the Kostal Piko in your application
Thank you in advance

Larnak23, 2019-07-26:
Hello perhaps some infos 

https://github.com/zevero/kostal

that may be used for this other solution
Thanks

DiedB, 2019-07-27:
Seems to be possible. Will look into it once I find some time, but that may take a while unfortunately.

Larnak23, 2019-07-27:
Thanksa lot 
It will be a good chrismas gift....

Meilleures salutations
Manuel Arango

DiedB, 2019-07-27:
Haha, it probably won't take that long :)

Larnak23, 2019-11-04:
Hello
any news..

Larnak23, 2019-12-21:
Hello 
Did you have any chance to look at my request ?
best regards

closed 2019-12-21 by Larnak23

DiedB, 2020-01-03:
Have you tested this script? Does it work for your inverter?

FalcoRutten, 2020-06-19:
Hi, I also have a piko convertor, and am willing to test/support.

DiedB, 2020-06-28:
@FalcoRutten Please track issue #92