DiedB/Homey-SolarPanels #71: Could not login to growatt server

xniemeij, 2019-07-21:
I have troubles to login at the growatt server 
On my desktop it isn’t a problem . In the homey app it keeps saying it can’t connect.  I already have changed my password to a 3 digits word

pro-sumer, 2019-07-21:
> I have troubles to login at the growatt server 

Might be a duplicate of #65.

Which version of the Solar Panels App are you using?

xniemeij, 2019-07-21:
The last one 3.1.1

pro-sumer, 2019-07-21:
> The last one 3.1.1

That’s the one with the login bug...

See https://community.athom.com/t/solar-panels-growatt-omvormer-aanmelden/13949/14 for a workaround.

@DiedB When will you release 3.1.2 with the fix?

DiedB, 2019-07-27:
Looks like something went wrong when upgrading 3.1.1 to 3.1.2 on the stable channel, that's a bug on Athom's side. I have re-submitted it as 3.1.3, directly to the stable channel.

pro-sumer, 2019-08-02:
@DiedB How long does this usually take?

I'm still on 3.1.1 and don't see any updates.

DiedB, 2019-08-02:
Strange, the update was in concept state. I have resubmitted it, should take no more than 3 days.

DiedB, 2019-08-02:
Has been published!

closed 2019-08-02 by DiedB