DiedB/Homey-SolarPanels #73: Feature request Inverter temperature visibility

jeroenvano, 2019-07-27:
During warm days it would be useful to keep an check on the current Inverter temperature. Is there a possibility to create a Homey flow trigger for temperature has changed? 

For example I use a Solar Edge Inverter that decreases in production above 50°C internal temperature. During the past few days in NL outside temp reached >40°C and the Inverter went above 50°C. A trigger in homey could activate an additional ventilator to cool the Inverter externally.

Regards,
Jeroen

DiedB, 2019-07-27:
Interesting use case. Do you know if SolarEdge supports this in their API? Can you see the inverter temperature in their app for example?

jeroenvano, 2019-07-27:
Just checked, it does not show any where in the app. It only shows on the direct local LAN inverter status webpage. So it looks that you are not able to retrieve from the app webpage.

DiedB, 2019-07-27:
The API docs do show temperature, so it might be possible. I will look into this once I have the time.

jeroenvano, 2019-07-27:
Thanks, let me know if you need some testing

DiedB, 2019-07-28:
Can you try accessing the following URL and see if temperature data is available for your inverter?

Fill in the site ID and the API key as you have used them for the Homey app. The inverter serial number can be found in the online portal or on the inverter itself (probably), it has the format XXXXXXXX-XX.

https://monitoringapi.solaredge.com/equipment/[siteId]/[inverterSn]/data?api_key=[apiKey]&startTime=2019-07-28%2000:00:00&endTime=2019-07-28%2023:59:59

jeroenvano, 2019-07-28:
Yes it does show a temperature value.

<temperature>40.7308</temperature>

DiedB, 2019-07-29:
Thanks! Do you see one temperature value per phase (3 in total) or one value on the whole page? I'm trying to figure out whether this also works for single-phase inverters.

jeroenvano, 2019-07-29:
One temp per Inverter so not per phase.

See below example for 1 Telemetry count:

This XML file does not appear to have any style information associated with
it. The document tree is shown below.
<data>
<count>1</count>
<telemetries>
<threePhaseInverterTelemetry>
<date>2019-07-28 23:16:39</date>
<totalActivePower>0.0</totalActivePower>
<powerLimit>0.0</powerLimit>
<totalEnergy>672825.0</totalEnergy>
<temperature>0.0</temperature>
<inverterMode>SLEEPING</inverterMode>
<operationMode>0</operationMode>
<L1Data>
<acCurrent>0.0</acCurrent>
<acVoltage>226.516</acVoltage>
<acFrequency>49.9948</acFrequency>
<apparentPower>0.0</apparentPower>
<activePower>0.0</activePower>
<reactivePower>0.0</reactivePower>
<cosPhi>0.0</cosPhi>
</L1Data>
<vL1To2>392.688</vL1To2>
<vL2To3>393.656</vL2To3>
<vL3To1>393.156</vL3To1>
<L2Data>
<acCurrent>0.0</acCurrent>
<acVoltage>226.906</acVoltage>
<acFrequency>49.9942</acFrequency>
<apparentPower>0.0</apparentPower>
<activePower>0.0</activePower>
<reactivePower>0.0</reactivePower>
<cosPhi>0.0</cosPhi>
</L2Data>
<L3Data>
<acCurrent>0.0</acCurrent>
<acVoltage>227.219</acVoltage>
<acFrequency>49.9947</acFrequency>
<apparentPower>0.0</apparentPower>
<activePower>0.0</activePower>
<reactivePower>0.0</reactivePower>
<cosPhi>0.0</cosPhi>
</L3Data>
</threePhaseInverterTelemetry>
</telemetries>
</data>

mikedebock, 2020-07-01:
@jeroenvano I created a pull request with SolarEdge temperature measurement.

DiedB, 2020-07-02:
@mikedebock Thanks a lot for the addition!

@jeroenvano Please try it at https://homey.app/nl-nl/app/it.diederik.solar/Zonnepanelen/test/

jeroenvano, 2020-07-02:
> @mikedebock Thanks a lot for the addition!
> 
> @jeroenvano Please try it at https://homey.app/nl-nl/app/it.diederik.solar/Zonnepanelen/test/

Thanks guys, trying it now. 
The app now shows the inverter temperature, nice!  
The app does not support to trigger a flow on temperature change.

jeroenvano, 2020-07-02:
The 15min. Temp update intervals also work nicely

mikedebock, 2020-07-02:
@jeroenvano the trigger for temperature change would be a nice one, then the data is really useful. A good enhancement for the next release ;)

DiedB, 2020-07-02:
I have pushed a fix for that (v4.0.15). It now uses the predefined `measure_temperature` capability shipped by Athom with a custom title to make clear that it is the inverter's temperature. Flow should automatically work.

mikedebock, 2020-07-02:
The trigger works like a charm!

jeroenvano, 2020-07-02:
> The trigger works like a charm!

Same here, nice work! Thanks guys!

jeroenvano, 2020-07-14:
Had no issues, temp measurements are working incl. triggers so imho this can go to production release.

DiedB, 2020-09-18:
Has been released to production!