DiedB/Homey-SolarPanels #76: Enlighten

wico1975, 2019-08-26:
Met het installeren op de Homey van de app Zonnepanelen is goed gegaan. Alleen met het toevoegen van de enphase installatie gaat het niet goed. Alles in gevuld krijg met inloggen fout melding error: object Object wat doe ik fout

TerryBouhuizen, 2019-08-26:
I’ve got the same issue here. Created a dev account at Enphase, filled in my user ID and API key and got the error: could not find that PairSession

DiedB, 2019-08-27:
Sounds like two different issues. @TerryBouhuizen can you try it again and report back if you get the same result? Do you have the latest app version?

If it doesn't work, please send me all data you are entering in the pairing session and I will try to debug it. diederik@outlook.com

TerryBouhuizen, 2019-08-27:
Just to get things straight:
I use both the API and username of the dev account, right?
Terry
    	
    	

    	Outlook voor iOS downloaden

TerryBouhuizen, 2019-08-28:
> You need the API key of the dev-account and the User ID of the Enlighten account, according to Enphase documentation.
>
> https://developer.enphase.com/docs/quickstart.html
>
> Please report back. If you get it working, please make a write-up of the steps. Helps future users!

Please find the screenshots for the data. I used both as you said 
    	
    	

    	Outlook voor iOS downloaden

TerryBouhuizen, 2019-08-28:
> Outlook voor iOS [3] downloaden

Ok! Got it working now, here's how: 

1. Create the developers account via https://developer.enphase.com/
2. After activating the account, go to Applications and click on the
default app (you may have to choose a plan, just go for the EUR 0,- or
"Watt" plan), press "Regenerate" and then save. 
3. Copy the newly created API key and save it
4. Create an user account via https://enlighten.enphaseenergy.com [4]
5. After activating (if you have it installed via Eneco/Suniverse that
can take up to 1 week after installation) go to <<Your name>>,
Settings/Instellingen, API-Settings/API-Instellingen
6. Here is the tricky part. Now go to https://developer.enphase.com/
again, login again and go to Applications and click on the default app.
Then copy and go to the Authorization URL, keeping in mind that you are
still loged in as an user in step 5 and grant the application access.
After doing so, close the screen and go back to
https://enlighten.enphaseenergy.com [4], <<Your name>>, 
Settings/Instellingen and under API-Settings/API-Instellingen check the
blue line "Show applications that have API access / Toepassingen
weergeven die over API-toegang beschikken" and see if the default app as
showed under 2 has access. If so, all is good.
7. Copy the user id and save it
8. Go to Homey and install SolarPanels/Zonnepanelen. Click Enphase and
fill in the API key and user id and press connect
9. You are done!

Regards,

Terry 

 

Links:
------
[1]
https://github.com/DiedB/Homey-SolarPanels/issues/76?email_source=notifications&amp;email_token=ANAGVB5C6T7ILXPGUTVE7UTQGZ6THA5CNFSM4IPRGJC2YY3PNVWWK3TUL52HS4DFVREXG43VMVBW63LNMVXHJKTDN5WW2ZLOORPWSZGOD5LFSIA#issuecomment-525752608
[2]
https://github.com/
[3] https://aka.ms/o0ukef
[4] https://enlighten.enphaseenergy.com/

DiedB, 2019-08-28:
You need the API key of the dev-account and the User ID of the Enlighten account, according to Enphase documentation.

https://developer.enphase.com/docs/quickstart.html

Please report back. If you get it working, please make a write-up of the steps. Helps future users!

wico1975, 2019-08-28:
Nee nog niet.
Ik moet denk ik even langs mijn installateur ik kom er niet in.

Verstuurd vanaf mijn iPhone

> Op 28 aug. 2019 om 19:22 heeft Diederik Bakker <notifications@github.com> het volgende geschreven:
> 
> Thanks a lot for the write-up! @wico1975 does it work for you as well, after using the guide above?

DiedB, 2019-08-28:
Thanks a lot for the write-up! @wico1975 does it work for you as well, after using the guide above?

DiedB, 2019-09-01:
Let me know if you need further help. I will close this issue as it's not app-related.

closed 2019-09-01 by DiedB