DiedB/Homey-SolarPanels #77: Solarman Timed out after 300000ms

chrisboer, 2019-08-30:
Updated the app to 3.1.3 as I still had a unregistered development install.

Now I get the message when I want to add the Solarman system 
"Timed out after 300000ms"

I can connect via web interface towards 
http://home.solarman.cn/main.html

But I cannot add a new appliance in Homey.

DiedB, 2019-09-01:
Are you sure it's not temporary downtime (these APIs are very unstable)? If not, please send your credentials to diederik@outlook.com and I will troubleshoot!

chrisboer, 2019-09-11:
Any update on the issue... I have send you my credentials 10 days ago.

DiedB, 2019-09-11:
I have received your email, but am very busy with other things in life so it might take a few weeks.

chrisboer, 2019-09-11:
Ok geen probleem just checking !

bedankt alvast als ik moet helpen hoor ik het wel.

Met vriendelijke groet,

Chris

chrisboer, 2019-09-17:
@DiedB it works again... so probably a temp API failure

closed 2019-09-17 by chrisboer