DiedB/Homey-SolarPanels #78: huawei omvormers

JeanPaul85, 2019-09-01:
Is het mogelijk om ook de huawei omvormer toe te voegen? Alvast bedankt voor de moeite.

DiedB, 2019-09-03:
What are you using to monitor your inverter currently?

JeanPaul85, 2019-09-03:
Een app op de telefoon en via een website op de laptop. https://eu.fusionsolar.huawei.com

DiedB, 2019-09-11:
Can you send your credentials to diederik@outlook.com so I can have a look?

Fire69, 2019-10-21:
I'm currently looking to have solar panels installed and Huawei is one if the proposed inverters.
@DiedB, have you had the possibility to look at @JeanPaul85's installation?

Fire69, 2019-10-21:
This topic might be helpful:
https://forum.pvoutput.org/t/add-batch-status-results-not-appearing/640/8

It has links to some pdf's with info about the API.

DiedB, 2020-01-03:
Do you still need this? I would need direct access to a Huawei inverter to implement this (maybe someone could port forward it in their router).

Fire69, 2020-01-03:
I would certainly like support for the Huawei. But my installation is only in February, so I can't give you access yet.

DiedB, 2020-01-03:
Keep me posted!

Fire69, 2020-03-10:
My panels finally got installed, with Huawei inverter.
What access do you need?  I don't know if direct acces is going to work.
There are 2 apps you can use:
- 1 connects locally to the inverter through the router or through the wifi-network of the inverter
- 1 connects to a webserver of Huawei

I tried connecting from work to home by VPN but then the app doesn't detect the inverter.

DiedB, 2020-03-10:
Is the app called FusionSolar?

https://forum.huawei.com/enterprise/en/communicate-with-fusionsolar-through-an-openapi-account/thread/591478-100027

Looks like they have an API, you need to request an account for that. Could you try that?

Fire69, 2020-03-11:
That's good news!
I mailed them to create an account for me.

Fire69, 2020-03-16:
I received credentials for my OpenAPI account.  No clue what to do with them.
Can I give them to you in some non-public way?

DiedB, 2020-03-16:
Sure! Send them to diederik@outlook.com

Fire69, 2020-03-19:
Did you get my mail?  No pressure, just making sure I didn't end up in your junk-mail :)

DiedB, 2020-03-19:
I did! It's on my to do list but might take a while before I find the time for it.

DiedB, 2020-03-21:
Hmm, does not seem to be working.

[image]
Cannot find any reference to that error code.

Can you locally connect to the inverter?

Fire69, 2020-03-21:
Yes, there is an app FusionHome (on Android) that lets you connect locally to the inverter.

DiedB, 2020-03-23:
Can you ask them what this error code means?

Fire69, 2020-03-23:
Done!

Fire69, 2020-03-23:
They tested with my credentials and say it works.
[image]

They sent a pdf how to connect to their service, but I think it's the same procedure as the link you posted earlier:

[2019FS026 Communicate with FusionSolar through an openAPI account.pdf](https://github.com/DiedB/Homey-SolarPanels/files/4371698/2019FS026.Communicate.with.FusionSolar.through.an.openAPI.account.pdf)

DiedB, 2020-03-24:
Apparently it needs to be in the POST body as well as the query parameters. Strange implementation, but it is indeed working. Will look into it later this week.

Fire69, 2020-03-24:
Great news!  Thanks for your time and effort! 💯

quahfamili, 2020-05-22:
@DiedB am i right to say that Huawei inverter is supported? my system is going to be installed in Jun. Would want to test it out too

DiedB, 2020-05-28:
I'm a little on the edge on supporting this. The thing is, everyone who wants to use it needs to send an email to Huawei, which severely limits the userbase of the driver. There is no good place to put an explanation on how to connect your Homey to a Huawei inverter, leading to lots of questions and confusion which costs time, in the end.

quahfamili, 2020-07-12:
hope you can get this supported. This is quite a good inverter afterall

Fire69, 2020-07-12:
I'm also still hoping for support, but I get that there is no easy way to do / support / explain it.
Huawei's implementation isn't very user friendly.  There isn't even a way to reset your password.
Think I might mail them to ask if this is going to change in the future.

DiedB, 2020-07-14:
The inverter might be good, but the connectivity is far from user friendly at this point. Implementing it would mean lots of support requests and would result in a very small user base.

I think I'll wait until they have their monitoring a bit more accessible. This issue will remain open in the meantime.

gertsve9, 2020-08-11:
I also have got an Huawei solar inverter. I look forward to this issue.

reyntjensw, 2021-02-25:
@DiedB Did anything change yet in the way the api works?
What are you missing or what are you looking for? Maybe we can create a case at Huawei for this.

DiedB, 2021-03-01:
@reyntjensw Not that I know of.

What I would like is a user friendly pairing process for the app (if each user has to send an e-mail to Huawei, it requires documentation and a lot of hand-holding for most users), and a well-documented and supported API.

reyntjensw, 2021-03-08:
@DiedB I understand, I can ask but I'm afraid it will not that easy to fix.
In the meantime, is there any way we can use it right now?

DiedB, 2021-03-08:
Maybe someone can record local traffic between the FusionSolar app and the inverter? Does the inverter advertise over mDNS?

I found [this](https://gitlab.com/Emilv2/huawei-solar) project that locally connects to the inverter over ModBUS. Can someone try it? Doing an `nmap` scan on the inverter and posting the results here would also help.

Fire69, 2021-03-08:
Anything you can learn from this Home Assistant integration?
https://github.com/Emilv2/huawei_solar
Seems like it reads the data from the inverter directly over ip instead of needing ModBUS.

DiedB, 2021-03-08:
It uses the repository I linked as a package, so I think that is still ModBUS? Correct me if I’m wrong.

Fire69, 2021-03-08:
Ah, didn't notice the repository :)

As far as I have found (info is pretty scarce) you can connect via ModBUS TCP.  Some say you have to connect to the built-in AP, others managed to connect to it over the router.

I'm going to mail Huawei to ask more info.

Fire69, 2021-03-08:
Mail sent.

Maybe you can get some more info from this topic, topic starter seems to have successfully connected over wifi.

quahfamili, 2021-03-08:
Really hope there is some head way to this. Recently I had a trip that is gone 2 weeks unnoticed.

Fire69, 2021-03-08:
Got a reply, not really the answer I was hoping for...
If I understand correctly, this is to completely replace their monitoring system with your own.  Don't really want that.
[2019INVL044 How to connect SUN2000L trough ModBus TCP to a Third-party monitoring.pdf](https://github.com/DiedB/Homey-SolarPanels/files/6104558/2019INVL044.How.to.connect.SUN2000L.trough.ModBus.TCP.to.a.Third-party.monitoring.pdf)

After searching some more, I did find a document describing their ModBUS implementation:
[SUN2000L MODBUS Interface Definitions.pdf](https://github.com/DiedB/Homey-SolarPanels/files/6104569/SUN2000L.MODBUS.Interface.Definitions.pdf)

DiedB, 2021-03-09:
Thanks for the effort! I think this is fine and should be enough to build a Homey integration. Could you forward port 502 of your Huawei inverter and contact me directly on the community forum, so I can have a look?

Fire69, 2021-03-09:
Done!

gertsve9, 2021-03-09:
Hi, Looks forward to this integration. Let me know if You want more inventers to test.

Med vänliga hälsningar

Gert Svensson

reyntjensw, 2021-03-14:
@DiedB just fyi I did find this repo of an integration working with the api keys (https://gitlab.com/martinstannard/solar).
Do you need some extra test setups so you can query other plants?

Fire69, 2021-03-14:
> @DiedB just fyi I did find this repo of an integration working with the api keys (https://gitlab.com/martinstannard/solar).
> Do you need some extra test setups so you can query other plants?

I tried getting this running on Ubuntu 2 months ago, didn't work.  Created an issue on github, no reply.
I'm now trying to get it running on Windows, this is even more frustrating... :/

DiedB, 2021-03-16:
I tried the Python library (which uses ModBUS) last week, that seems to work well. However, there are some timing-related quirks with the Huawei implementation which I need to look at.

Fire69, 2021-03-16:
> I tried the Python library (which uses ModBUS) last week, that seems to work well. However, there are some timing-related quirks with the Huawei implementation which I need to look at.

Does this help?
https://forum.huawei.com/enterprise/en/forum.php?mod=redirect&goto=findpost&ptid=603386&pid=3364703

DiedB, 2021-03-16:
Yeah, that's what I found as well. The issue was the client ID, most libraries default to `1` while the inverter expects `0`. My PoC works after changing that, will add it to the Homey app tomorrow if I find the time.

DiedB, 2021-03-17:
Could you all test the version at https://homey.app/a/it.diederik.solar/test/?

And can someone check whether the FusionSolar app shows the inverter IP address anywhere? If that's the case, could you make a screenshot of that screen and document the steps taken to get to that screen, so that I can add it to the app documentation?

Fire69, 2021-03-17:
Installed the app, added the inverter, worked flawlessly! 💯 

Can't find the IP anywhere in the FusionSolar app unfortunately, I guess the user will need to check it in the router.

reyntjensw, 2021-03-17:
@Fire69 did you have to enable something else?
I'm getting an ECONNREFUSED.

I'm using a SUN2000-6KTL

Fire69, 2021-03-17:
@reyntjensw no, didn't have to do anything special to make it work.
Where do you get that error?  While adding the device to Homey?

DiedB, 2021-03-17:
@reyntjensw Maybe the ModBUS interface of your inverter has to be enabled in the FusionSolar app? Or the IP does not match?

reyntjensw, 2021-03-17:
@Fire69 @DiedB after adding the IP I'm getting the connection error.
I have been reading that the modbus is only working on the wifi antenna, not on systems that work with an ethernet cable.

I think I already have the correct ip, but just to be sure, I've created a case at Huawei.
Once I have more info, I'll post it here

gertsve9, 2021-03-18:
Same problem for med, tried with correct IP. 
I'm getting an ECONNREFUSED.
Wired connection.

DiedB, 2021-03-19:
Apparently port 502 is closed on wired connections? Can someone do a port scan to see what ports are open?

gertsve9, 2021-03-20:
I find no open ports at all. 

/Gert Svensson

Fire69, 2021-03-20:
On wifi only 502 is open.

How are you guys connecting the inverter by cable?  Normally it only has wifi.

DiedB, 2021-03-20:
I am also a little confused. Do you use a Smart Dongle-WLAN-FE? That one should support both WiFi and Ethernet connections. The inverter itself should also have a RJ45 port, but that one is meant for RS485 communications.

gertsve9, 2021-03-20:
Yes I have that dongle, maybe the wifi is disabled since I dont use it. I will check it.

Vänligen

Gert Svensson

Mail ***@***.***

DiedB, 2021-03-20:
I found [this](https://forum.huawei.com/enterprise/en/sun2000-smart-dongle-modbus-tcp-issues/thread/675885-100027?page=1) thread describing the issues. Port 502 is closed when using an ethernet connection but works fine over WiFi.

Best workaround currently is connecting the inverter to your WiFi network. Could you try that?

Fire69, 2021-03-20:
Try mailing eu_inverter_support@huawei.com to see if you can open the port on that adapter.  Huawei support is pretty good/fast.

[60 more replies]

closed 2022-03-17 by DiedB