DiedB/Homey-SolarPanels #79: Feature Request Enphase Metered

Pirke123, 2019-09-07:
I have a 3 phase Enphase metered system that measures per phase the production of the solar panels as well as per phase the consumption from the grid (or production towards the grid). The difference is the total consumption per phase in my house.

Is it possible to have these statistics (preferably per phase) available in Homey as well?

DiedB, 2019-09-11:
Hi! I think that would be possible. However, I do not have much time on my hands currently. I will keep the issue open, and it might be implemented in the future. Meanwhile, if you read this and want this feature as well, please indicate so in this issue thread.

DiedB, 2020-07-23:
See #110

closed 2020-07-23 by DiedB