DiedB/Homey-SolarPanels #80: Class 'solarpanel' V3 firmware

jaxc0402, 2019-09-16:
with the new v3 update of Athom there is a new "class": "solarpanel", that places the solar panels in the right class.
In you’re app this needs an update. So in v3 all works well.

This is needed for the new 'energy' function of V3 homey. Could this plz be added??

DiedB, 2019-09-16:
I found it in the developer documentation as well, will definitely be implemented.

Do you know if there is any way to run V3 on a Homey currently?

jaxc0402, 2019-09-16:
I am in AATP (beta tester) and have v3 running. I have send you a pb via Slack (easier to communicate ;) )
You could ask Athom to join AATP to ??

pro-sumer, 2019-09-16:
I hope this migration is possible without data loss (without losing historic production data).

DiedB, 2019-09-16:
It should be, by calling `Device.setClass('solarpanel')` on existing devices.

pro-sumer, 2019-09-16:
OK!

Where can I read about this new API?

Reason for asking: I use a fork of the smart meter (P1) App and wonder whether I need to update anything there.

pro-sumer, 2019-10-15:
Please have a look at this:

https://github.com/koktaildotcom/com.p1.smartmeter/pull/9#issuecomment-542118623

Would be awesome if you can add this in such a way that we don't loose historic data!

pro-sumer, 2019-10-15:
> Where can I read about this new API?

For anyone interested: https://apps.developer.athom.com/tutorial-Drivers-Energy.html

Also see: https://blog.athom.com/developers-homey-energy/

DiedB, 2019-10-15:
See 765ddebbcdddfbd48b5b9e434cb7a86ae8e7aa0d. I'm afraid migration will result in data loss, although we could probably keep old data in a different log. I will look further into this, but the SDK limits the possibilities here.

DiedB, 2019-10-22:
Will be pushed to the beta channel tomorrow (most likely).

closed 2019-10-22 by DiedB

pro-sumer, 2019-10-29:
Athom released v3 today.

@DiedB Can you promote the App to stable? (or are there still issues to be solved?)

(I'd rather not change to the beta channel, as I would loose all my historic production data)

DiedB, 2019-10-29:
@pro-sumer There are currently 15 beta users, don't think that is enough for promoting the app to stable. Additionally, the current update would erase everyone's production data (although I don't think many users value data retention, only live production stats). There is currently no solution for that.

pro-sumer, 2019-10-29:
Does `athom app install` in the beta branch (instead of using the beta channel) also lead to data loss?

I just updated (my private fork of) the P1 App and that did keep historic data (but that was already side loaded before the update)

What exactly triggers the data deletion?