DiedB/Homey-SolarPanels #82: App paused without notice

Raycokes2000, 2019-10-13:
[image]
App was paused after restart it gave error like screenshot

Raycokes2000, 2019-10-13:
Strange can not add screenshot?

DiedB, 2019-10-13:
I got the screenshot in my mail, but not here (for the better, as it publishes your API key). Seems like a one-off issue, as your Homey was not yet connected to Wi-Fi just after it booted up. Strange that it crashed the app in this way, that should not be happening. Did functionality restore automatically after unpausing the app?

Raycokes2000, 2019-10-14:
Hmm OK, it is working again. Probably like you said, after restart reconnection problem. Still strange that app got paused in the first place

DiedB, 2020-01-03:
Did it ever happen since? Please re-open the issue if that is the case.

closed 2020-01-03 by DiedB