DiedB/Homey-SolarPanels #83: Feature request changing update interval

RMVolkers, 2019-10-21:
Solaredge is updating every five minutes. Now the intervals are 20min and 10min in the app. It would be nice if you can change the time interval yourself in the menu.

DiedB, 2019-10-21:
If I recall correctly, SolarEdge does not allow more than 10 API requests per hour, spread out over the entire day. The app would hit rate limits if the frequency would be increased. Please let me know if this has changed.

RMVolkers, 2019-10-21:
This is the info i have (Jan 2019):

Daily Limitation
Use of the monitoring server API is subject to a query limit of 300 requests for a specific account token and a parallel query limit of 300 requests for each specific site ID from the same source IP.
APIs that do not have a specific ID (e.g. Site List, Account List) will be counted as part of the account query limit. Any additional site or account level request will result in HTTP 429 error – too many requests.
Concurrency Limitation
The monitoring server API allows up to 3 concurrent API calls from the same source IP. Any additional concurrent calls will return HTTP 429 error – too many requests.
To execute APIs concurrently without exceeding the above limitation, it is the client responsibility to implement a throttling mechanism on the client side.
Bulk Use
Some of the APIs offer a bulk form, that is, they take multiple site IDs as an input. The list is separated by a comma and should contain up to 100 site IDs.
A bulk call for multiple sites consumes 1 call from the daily quota allowed for each of the sites included in the call.

DiedB, 2019-10-21:
Seems like that policy has not changed...

closed 2019-10-22 by DiedB