DiedB/Homey-SolarPanels #85: v4.0.0 Confusing tags in flows

pro-sumer, 2019-10-31:
Sideloaded v4.0.0 and noticed that my solar panel flow is no longer working, which should not have been a surprise as I see this in the timeline: "_The Solar Panels App has been migrated to Homey v3.0. Please check your flows_".

My flow sends me the daily production via a push notification, 15 minutes after sunset. The tag for the value has been removed during the upgrade. When trying to fix this I got confused by the tags; there are two for the SolarPanels App:
* _Energie_ (probably the current production, as it is 0 at the time of submitting this issue, after sunset)
* _Energiemeter_ (probably the daily production)

I don't see the string "Energiemeter" in your `app.json`, so I'm not sure whether this is a bug in the SolarPanels App or the Homey App itself.

What do you think? Is it possible to improve this?

pro-sumer, 2019-10-31:
Oh, I did not notice this before, but the same two strings are used in the main screen of the App as well:
* 0 W Energie
* X kWh Energiemeter

DiedB, 2019-11-04:
In order to support Homey Energy, I have to use their built-in capabilities (`measure_power` and `meter_power`). These result in incorrect descriptions for production and power values. In the past, I solved this by using my own capabilities, but this doesn't work with Homey v3.0. This is out of my hands :(

You could suggest Athom to change the display strings for these capabilities when the device has the `solarpanel` class?

pro-sumer, 2019-11-04:
What’s the best way to contact them about this?

pro-sumer, 2019-11-05:
A colleague uses the [SMA Modbus App](https://github.com/jghaanstra/sma.modbus) for his solar panels.

This App is also updated for Homey Energy, but it does show correct labels.

Would inspecting their `app.json` help?

pro-sumer, 2019-11-05:
They seem to add this to the `drivers:` section:

```
      "capabilitiesOptions": {
        "meter_power": {
          "title": {
            "en": "Daily Yield",
            "nl": "Dagopbrengst"
          }
        }
      },
```

However, this does not seem to work in SolarPanels?

DiedB, 2019-11-06:
Think it should work but has to be applied to every driver. Will look into this soon.

phirk, 2019-11-06:
> Think it should work but has to be applied to every driver. Will look into this soon.

Ah I missed that you need to do this *per driver*.

Tried it for my Growatt inverter and it's working indeed!

DiedB, 2020-01-03:
Have pointed it out to Athom as I believe this is something that should be fixed in Homey software instead of overriding it here.

closed 2020-05-16 by DiedB