DiedB/Homey-SolarPanels #86: Virtual inverter/solar plant

reine73, 2019-11-02:
Would it be possible to add a virtual inverter/solar plant were the values could be set through flows?
Would it be 

That could open up for a few interesting features usecases like monitor own usage and sold electricity for example as well as being used if you have an unsupported inverter but can get the data in other ways.
 Could also be used for small scale wind power as well.

I currently doesn't have any solarpanels but are in the process of installing and will most likelly use a EnergyHub by FerroAmp for my installation wich is not supported.

DiedB, 2019-11-04:
Why do you want a virtual device (which is a little hacky in my opinion), and not just EnergyHub support in this app or in its own app?

reine73, 2019-11-05:
Native support for EnergyHub would of course be perfect in my case.
But the whole thought was that it would also open up for new of possibilities for experimenting and also would be able to integrate with different systems much faster if there wasn't a real connection towards the API's of the specific company/inverter.

Obviously, if you think it would be a hacky/bad solution, you shouldn't implement it. It was suggestion based on that I installed it and wanted to experiment and thought of possibilities... :)

closed 2022-01-22 by DiedB