DiedB/Homey-SolarPanels #87: Display sunrise and sunset in App

pro-sumer, 2019-12-21:
Since solar panels can only produce power between sunrise and sunset it would be nice if the timestamp of both events can be displayed, for the current location of the panels (this probably requires the user to enter longitude and latitude of that location during setup).

For example:

|Column 1|Column 2|
|:-:|:-:|
|*1200 W*|*1.1 kWh*|
|Power production|Energy generation today|
|*8:30*|*17:30*|
|Sunrise|Sunset|

You can use this API for obtaining the timestamps: https://sunrise-sunset.org/api

pro-sumer, 2019-12-30:
Made a mock-up (with random hardcoded strings) to clarify what I would like:

[image]

(for this example I used icons from [Nikita Golubev](https://www.flaticon.com/authors/nikita-golubev))

DiedB, 2020-01-03:
Thanks for the idea and the mock-up! I don't think this is functionality that should be added to the Solar Panels app in Homey as it is not directly related to the device itself. This Homey app should not become a replacement for the app of your inverter brand for example. Its functionality should be limited to providing data that is relevant for connecting your inverter to other smart home devices through Homey. Providing sunrise and sunset data is something that is already implemented by Athom in that regard.

Timestamps for when production started and finished today might be relevant to the device itself, but I would need a use case for that before spending time on the implementation. The latter is pretty hard to implement by the way :smile:

pro-sumer, 2020-01-03:
> This Homey app should not become a replacement for the app of your inverter brand for example. 

Ah, good to know that's the design philosophy (I was indeed more hoping for the opposite: not needing the vendor specific, ugly, App).

DiedB, 2020-01-03:
Would be nice, but Homey is too constrained to make a good app for that - a better route would then be to make an open source native app that talks to the API of your inverter manufacturer.

closed 2020-01-03 by DiedB