DiedB/Homey-SolarPanels #88: Display total/yearly production in App

pro-sumer, 2019-12-21:
While it's nice to see the current and daily production numbers, it is also useful to see the total and yearly production numbers (if available).

It would be great if you could add that for inverters that support this.

For example the Growatt API provides the total production, but not the yearly. However, maybe the latter can be calculated by letting the user enter the total production of last year (once / every year), if (s)he knows?

pro-sumer, 2019-12-30:
Made a mock-up (with random hardcoded strings) to clarify what I would like:

[image]

(for this example I used icons from [Ctrlastudio](https://www.flaticon.com/authors/ctrlastudio))

DiedB, 2020-01-03:
Looks good, will consider implementing it. I'm not a fan of manually entering/calculating yearly values, but some manufacturers have them available I believe.

locksen-nl, 2021-04-18:
@DiedB Tried to add it the function my self but im not skilled enough  I need to add yieldtotal for the Solax inverter

HenkN, 2021-08-10:
Would be great to have this feature. Especially the total kWh since start. SolarEdge supports this value.

This option is also needed when using the new Energy features of Home Assistant.

mhoogenbosch, 2021-11-06:
There is a Power by the Hour app too. This provides this information just like suggested. Just a thought.

DiedB, 2021-11-09:
@HenkN The beta version exposes the total production for SolarEdge systems.

I will add the values to the app where APIs supply them. However, I'm not going to infer the total energy produced from power values. For that, the Power by the Hour app suffices.

closed 2021-11-09 by DiedB