DiedB/Homey-SolarPanels #91: Error login Growatt

Kingster2, 2020-01-14:
Since a few weeks the Growatt login gives a error:

Apparaat niet beschikbaar
Error retreiving data (FetchError: request to https://server.growatt.com/newLogimAPI.do failed, reason: getaddrinfo EAI_AGAIN server.growatt.com:443)

Password and login have been checked multiple times and are correct.

DiedB, 2020-01-15:
@pro-sumer Could you take a look?

pro-sumer, 2020-01-15:
Actually I had a similar error yesterday...

@Kingster2 I write "similar" because the URL in the first post is incorrect (`Logim` should be `Login`). Did you manually type this URL and made a typo? Or was the Homey really trying that URL?

For me this problem occurred due to DNS issues on my network. After solving those most domain name lookups worked again, except for this particular one. Even restarting the Solar Panels App did not help. Restarting the Homey device did help though. Did you already try restarting the Homey? If not, can you please try that first?

Kingster2, 2020-01-15:
This was a typo, sorry for that. Couldn't copy/paste from the Homey screen.
Restarting the Homey did the job. Thank you for your help.

closed 2020-02-20 by DiedB