DiedB/Homey-SolarPanels #92: Kostal support

carstenkruseh, 2020-01-28:
Hi!

Would it be possible to add Kostal support for the Homey app?

I have a Plenticore Plus - It's available on my LAN.
I Hope you can help :)

DiedB, 2020-01-29:
Can you show some screenshots of the LAN interface?

carstenkruseh, 2020-01-29:
Hi!

Thanks a lot for your reply!

First login-page:

Maybe possible to Connect to sunspec? Or simular?

Sendt fra min iPhone

DiedB, 2020-01-29:
You would need to upload those images via GitHub, otherwise they are not shown.

carstenkruseh, 2020-01-30:
Hi!

Here is the pictures :)
Maybe you can use something from this: (Though it's for Kostal Pico-inverter)
https://www.smarthomeng.de/developer/plugins/kostal/README.html

[image]
[image]
[image]

DiedB, 2020-01-30:
Should be possible, I think. Could you forward port 80 of your inverter and send me the IP and login details? I will have a look.

diederik@outlook.com

FalcoRutten, 2020-06-29:
94.215.98.26
port 80
user pvserver
pw test123

derhelle, 2020-08-26:
Hi DiedB,

i also would appreciate an Kostal Integration. I could send you the requested Data or give you access to my Inverter.
A Tesla PowerWall 2 integration would be also very cool ;-)

regards Helle

DiedB, 2020-08-27:
@derhelle Could you contact me at diederik@outlook.com? Would like to look into both of them, but port forwards of others here are not working (anymore).

DiedB, 2020-08-27:
Almost done, it just needs some error handling and a token refresh mechanism. Implementing the authentication is hell, the other parts are smooth sailing. I like the update speed of these inverters, a 5-second resolution is possible! Will upload a test version later tonight or tomorrow.

[image]

carstenkruseh, 2020-08-27:
Sounds super great!
Will it be possible to see battery %, house consumption too?

91.144.213.80
Cahan277

Hent Outlook til iOS<https://aka.ms/o0ukef>

[IMG_7333]<https://user-images.githubusercontent.com/7628001/91479323-8bbcb400-e8a1-11ea-8e2d-8599eae6fe8e.PNG>

derhelle, 2020-08-27:
Awesome!! Thank you!

DiedB, 2020-08-28:
@carstenkruseh Could you reset the inverter password? I locked the account while testing the login error handling ;-)

carstenkruseh, 2020-08-28:
Done - Password Now Cahan2777 :)

Hent Outlook til iOS<https://aka.ms/o0ukef>

DiedB, 2020-08-28:
Great, thanks for your collaboration guys!

You can test Kostal integration at:
https://homey.app/a/it.diederik.solar/test/

I would like to implement full support for your system @carstenkruseh, but it is a little too complicated for this app currently. For all brands it supports, it only shows PV production statistics. It would need some more custom capabilities and such, and I think it is better to create a standalone Kostal app which allows for full integration. Athom is releasing a new SDK (with firmware 5.0.0) and once that is adopted broadly, I'll plug Kostal out of this app and into its own app, including battery support. Could you create an issue for now (in this repository) asking for full support?

There is another limitation, since the app currently only supports PV/inverter variables, it won't show the values you currently want to see @carstenkruseh. The app measures inverter output and yield, but because you have a battery on the DC side, the inverter output doesn't equal solar production. Don't worry, we'll get around to it once Kostal gets its own app ;)

FalcoRutten, 2020-08-28:
Hi,
I tested the kostal version, but get timeout-error.
Falco

DiedB, 2020-08-28:
@FalcoRutten What are you entering exactly, can you send screenshots of your inverter interface?

FalcoRutten, 2020-08-28:
See attachments

Verzonden vanaf mijn Samsung Galaxy-smartphone.

DiedB, 2020-08-28:
@FalcoRutten You'll have to do that directly on GitHub, it doesn't allow email attachments to come through.

FalcoRutten, 2020-08-28:
[image]
[image]

DiedB, 2020-08-28:
Ah, you have a different inverter from @carstenkruseh and @derhelle (they have a Plenticore), and the app is not going to work with yours (for now). I'll need to clarify that in the app. Kostal does not really provide documentation on their inverters and connectivity, so it is hard to determine which inverters are supported.

FalcoRutten, 2020-08-28:
Would be great if you could also add my system!!
Thanks
Falco

derhelle, 2020-08-28:
Hi, I just tried but I get an error: Could not find that PairSession.

regards Helle

carstenkruseh, 2020-08-30:
It’s meassuring Home consumption, not pv production :)
Maybe you need to only add DC string 1 and 2.
String 3 is battery.

Hent Outlook til iOS<https://aka.ms/o0ukef>

DiedB, 2020-08-30:
@carstenkruseh Are you sure? I believe it is consistent with the inverter output, e.g. the value that you see on the top left when you login to your inverter. It might look like home consumption because the home often runs on battery power (which goes through the inverter).

carstenkruseh, 2020-08-30:
Hi!

Yes, I’m sure.
But you might be right.

I just tested this afternoon.
The consumption was bigger than the pv output.
So the rest of the power was drawn from the battery.

Pv output + battery output = consumption.
So you are probably right.

Then I’ll have to wait for your Kostal app :)

Med venlig hilsen
Carsten Kruse
Kristrupvej 33
8960 Randers SØ

T:	31 61 31 21	E:	carstenkh@gmail.com <mailto:carstenkh@gmail.com>

DiedB, 2020-09-18:
The newest version including Kostal Plenticore has been released.

closed 2020-09-18 by DiedB

FalcoRutten, 2020-09-18:
Hi, Still not working with my Piko Kostal!

madmax75, 2023-03-05:
Hi @DiedB, is it possible to include the Kostal Plenticore inverters in the app?