DiedB/Homey-SolarPanels #96: 4.0.6: Error retrieving data (TypeError: response.headers.getAll is not a function)

pro-sumer, 2020-02-20:
Today the App updated to 4.0.5 (which stopped working, but you got notified by Athom/Homey according to the error) and 4.0.6 (where I don't see that message) and now it's no longer working for me:

```
Error retrieving data (TypeError: response.headers.getAll is not a function)
```

I'm not sure whether that's specific for my device (Growatt) or all.

DiedB, 2020-02-20:
Whoops, `node-fetch` was upgraded and `headers.getAll()` seems deprecated in favour of `headers.get()`.

Could you test https://homey.app/nl-nl/app/it.diederik.solar/Zonnepanelen/test/ and verify whether it works again?

pro-sumer, 2020-02-20:
“Crashed during loading”

(And the developer has been notified)

pro-sumer, 2020-02-20:
I get this stack trace when installing version 4.0.6 using the Athom CLI:

```
module.js:550
    throw err;
    ^

Error: Cannot find module 'node-fetch'
    at Function.Module._resolveFilename (module.js:548:15)
    at Function.Module._load (module.js:475:25)
    at Module.require (module.js:597:17)
    at Module.require (/opt/homey-client/system/manager/ManagerApps/bootstrap/sdk/v2/index.js:1:253)
    at require (internal/module.js:11:18)
    at Object.<anonymous> (/drivers/enphase/api.js:1:77)
    at Module._compile (module.js:653:30)
    at Object.Module._extensions..js (module.js:664:10)
    at Module.load (module.js:566:32)
    at tryModuleLoad (module.js:506:12)

--- INFO: it.diederik.solar has been killed ---
```

This is my configuration:
* Homey firmware 4.0.0
* Homey iOS App 4.0.2.474

pro-sumer, 2020-02-20:
This delta in the Growatt `api.js` file seems to fix the error:

```
- this.cookies = response.headers.getAll('Set-Cookie').map(directive => directive.split(';')[0]).join(';')
+ this.cookies = response.headers.raw()['set-cookie'].map(directive => directive.split(';')[0]).join(';')
```

Note that I had to restart my Homey, after installing the fixed version using the "Homey CLI" app.

martijngeerts, 2020-02-22:
same problem here

DiedB, 2020-02-22:
Could you try the updated test version (v4.0.8)? Something went (horribly) wrong with the app's dependencies.

pro-sumer, 2020-02-22:
4.0.9 works for me.

martijngeerts, 2020-02-23:
4.0.9 works for me too, Now waiting for a little sunshine ;)
Thx for the quick reply

closed 2020-02-25 by DiedB