DiedB/Homey-SolarPanels #97: Ginlong - not capturing production data

kmelhus, 2020-02-27:
Device Ginlong showing a few watts while production is > 5000. I have checked that ID is correct. 
A bit concerned that I'm missing something. Grateful for any clue.

DiedB, 2020-03-05:
Strange, no idea what would cause that. Does the Ginlong iOS/Android app work?

kmelhus, 2020-03-05:
Thank you for responding. Yes, no problem with iOS app. Anything I can do to contribute?
Just now iOS app shows  3360w and your app 376,69w

DiedB, 2020-03-05:
Strange. Is the pattern consistent with the iOS data? Could the numbers in Homey be divided by 10, for example?

kmelhus, 2020-03-05:
I'll check and get back to you

kmelhus, 2020-03-05:
No, pattern is not consistent, now 6800 / 513,66. If Power meter is Daily production, it is more likely that it's from another plant. I have reinstalled the device x times and checked the ID.

DiedB, 2020-03-05:
That is very strange. Have you already tried to re-add the inverter?

DiedB, 2020-03-05:
Hmm that is very strange. Can you send me your credentials at diederik@outlook.com? I will try to look into the issue.

kmelhus, 2020-03-06:
Thanks and done :)

closed 2020-03-07 by DiedB